jopohl/urh #360: Playback does not work more than once with HackRF

Alex-111, 2017-11-16:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
Sending with HackRF should work more than one time...

## Actual Behavior
Sending with HackRF only works once. After sending a waveform again the progress bar does not appear and nothing seems to be sent. Resetting the HackRF sometimes helps to make it work again...

Unplugging HackRF and restarting URH always solves the issue. After that most of the times sending works only once again....

## Steps to Reproduce the Problem

  1. Record with e.g. 8Mhz
  2. Playback the recording
  3. Stop the playback and press the play button again -> No progress bar and sending does not work

## Platform Specifications

  - Python Version:
  - Operating System: Windows 10
  - Version of URH: 1.8.12
  - URH was installed:  via ```.msi```

jopohl, 2017-11-16:
What firmware version is your HackRF running?

Alex-111, 2017-11-17:
It runs on 2017.02.1

closed 2017-11-18 by jopohl

Alex-111, 2017-11-20:
Thanks a lot. Seems to work now...

jopohl, 2017-11-20:
You're welcome! That was a nasty one indeed.

Agmagor, 2017-11-23:
I don't think this issue is completely solved.
I attached to this message a screenshot showing that the hackrf is still active after pressing the stop button in the Send Signal window.
Pushing the hackrf reset button shuts it down properly.

The URH log says:

> [INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: Stopped receiving: Stop button clicked
[DEBUG::Device.py::device_send] wrappertype: exit requested. Stopping sending
[WARNING::Device.py::stop_tx_mode] HackRF: Transmit process is still alive, terminating it
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread

[image]

jopohl, 2017-11-23:
Does this happen everytime? Are you using the latest version of URH? This line here indicates what is wrong:
[WARNING::Device.py::stop_tx_mode] HackRF: Transmit process is still alive, terminating it

URH forcefully shuts down the HackRF process, because it blocks and therefore the HackRF stays in sending mode.

PR #361 (2017-11-18): Properly shut down hackrf device
fix #360