jopohl/urh #366: sdrplay Error problem

xykts, 2017-11-26:
windows7 x64     gnuradio_3.7.11_win64.msi 

Traceback (most recent call last):
  File "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py", line 23, in <module>
    from gnuradio import gr
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\gr\__init__.py", line 44, in <module>
    from top_block import *
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\gr\top_block.py", line 28, in <module>
    import gr_threading as _threading
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\gr\gr_threading.py", line 35, in <module>
    from threading import *
  File "C:\Program Files\GNURadio-3.7\gr-python27\lib\threading.py", line 13, in <module>
    from collections import deque as _deque
ImportError: Bad magic number in C:\Program Files\Universal Radio Hacker\collections\__init__.pyc

## Platform Specifications

  - Python Version:
  - Operating System:windows7 x64
  - Version of URH:1.8.13
  - URH was installed:Universal.Radio.Hacker-1.8.13-amd64.msi

jopohl, 2017-11-26:
As this is a GNU Radio related error: Is GNU radio the latest version? Can you use SDRPlay from GNU radio?

jopohl, 2017-11-26:
Just did some experiments on my Windows VM, and could reproduce this issue. Thanks for bringing this up! I just pushed a fix, which will be included in next release. For an instant fix, you can replace the content of file ``` C:\Program Files\Universal Radio Hacker\sdrplay_recv.py ``` with this:

```python
#!/usr/bin/env python2
##################################################
# GNU Radio Python Flow Graph
# Title: Top Block
# Generated: Fri Aug 21 15:56:13 2015
##################################################

from optparse import OptionParser

import tempfile
import os
import sys
try:
    with open(os.path.join(tempfile.gettempdir(), "gnuradio_path.txt"), "r") as f:
        gnuradio_path = f.read().strip()

    os.environ["PATH"] = os.path.join(gnuradio_path, "bin")
    # Append script path at end to prevent conflicts in case of frozen interpreter
    sys.path.append(sys.path.pop(0))
    sys.path.insert(0, os.path.join(gnuradio_path, "lib", "site-packages"))

except IOError:
    pass

from gnuradio import gr
from gnuradio.eng_option import eng_option
from grc_gnuradio import blks2 as grc_blks2
from InputHandlerThread import InputHandlerThread
import osmosdr
from gnuradio import zeromq

class top_block(gr.top_block):
    def __init__(self, samp_rate, freq, gain, bw, port):
        gr.top_block.__init__(self, "Top Block")

        ##################################################
        # Variables
        ##################################################
        self.samp_rate = samp_rate
        self.gain = gain
        self.freq = freq
        self.bw = bw

        ##################################################
        # Blocks
        ##################################################
        self.osmosdr_source_0 = osmosdr.source(args="numchan=" + str(1) + " " + "sdrplay")
        self.osmosdr_source_0.set_sample_rate(samp_rate)
        self.osmosdr_source_0.set_center_freq(freq, 0)
        self.osmosdr_source_0.set_freq_corr(0, 0)
        self.osmosdr_source_0.set_dc_offset_mode(0, 0)
        self.osmosdr_source_0.set_iq_balance_mode(0, 0)
        self.osmosdr_source_0.set_gain_mode(False, 0)
        self.osmosdr_source_0.set_gain(gain, 0)
        self.osmosdr_source_0.set_if_gain(gain, 0)
        self.osmosdr_source_0.set_bb_gain(gain, 0)
        self.osmosdr_source_0.set_antenna("", 0)
        self.osmosdr_source_0.set_bandwidth(bw, 0)

        self.zeromq_push_sink_0 = zeromq.push_sink(gr.sizeof_gr_complex, 1, 'tcp://127.0.0.1:' + str(port))

        ##################################################
        # Connections
        ##################################################
        self.connect((self.osmosdr_source_0, 0), (self.zeromq_push_sink_0, 0))

    def get_samp_rate(self):
        return self.samp_rate

    def set_samp_rate(self, samp_rate):
        self.samp_rate = samp_rate
        self.osmosdr_source_0.set_sample_rate(self.samp_rate)

    def get_gain(self):
        return self.gain

    def set_gain(self, gain):
        self.gain = gain
        self.osmosdr_source_0.set_gain(self.gain, 0)
        self.osmosdr_source_0.set_if_gain(self.gain, 0)
        self.osmosdr_source_0.set_bb_gain(self.gain, 0)

    def get_freq(self):
        return self.freq

    def set_freq(self, freq):
        self.freq = freq
        self.osmosdr_source_0.set_center_freq(self.freq, 0)

    def get_bw(self):
        return self.bw

    def set_bw(self, bw):
        self.bw = bw
        self.osmosdr_source_0.set_bandwidth(self.bw, 0)

if __name__ == '__main__':
    parser = OptionParser(option_class=eng_option, usage="%prog: [options]")
    parser.add_option("-s", "--samplerate", dest="samplerate", help="Sample Rate", default=100000)
    parser.add_option("-f", "--freq", dest="freq", help="Frequency", default=433000)
    parser.add_option("-g", "--gain", dest="gain", help="Gain", default=30)
    parser.add_option("-b", "--bandwidth", dest="bw", help="Bandwidth", default=200000)
    parser.add_option("-p", "--port", dest="port", help="Port", default=1337)
    (options, args) = parser.parse_args()
    tb = top_block(float(options.samplerate), float(options.freq), int(options.gain),
                   float(options.bw), int(options.port))
    iht = InputHandlerThread(tb)
    iht.start()
    tb.start()
    tb.wait()
```

jopohl, 2017-11-27:
New release is out. I will close here. Drop a comment if you still have trouble.

closed 2017-11-27 by jopohl