jopohl/urh #368: Global python error

nocomp, 2017-11-27:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
i use an usrp with gnu radio without any issue, everything works fine
when launching urh, it doesn t see my gnuradio install and i can modify the path either

## Actual Behavior
global python error
https://imgur.com/a/JJpo3

## Steps to Reproduce the Problem

  1.installed .msi version
  2.plugged usrp
  3.launched urh

## Platform Specifications

  - Python Version: 2.7.10
  - Operating System: win 10 64b
  - Version of URH: 1.8.14
  - URH was installed: __from  ```.msi```

jopohl, 2017-11-27:
I just pushed a new release (1.8.15) that should fix this issue.

nocomp, 2017-11-28:
hi johannes,
thank you for your update, no python error, but not working either.
i ve recorded a video for you, hope it can help us to understand what is not right
thxx for your time!
https://youtu.be/BI2I9EG4qlY

andynoack, 2017-11-28:
Hi @nocomp,
we have seen your video, thanks for that! We saw that you are using the native backend for the USRP, so gnuradio is not used in this case. Actually we are unsure about the USRP B series as we do not posses one of these devices. Next, you could try out the gnuradio backend for the USRP and see if this works. Make sure you use version 1.8.15 of URH for this.

Furthermore, you could share the content of ``` %TEMP%\urh.log ``` with us after pressing the record button.

nocomp, 2017-11-28:
hi, happy to help :)
sorry my english is not good enough i dont understand what you mean by "gnuradio backend"
give me a procedure, i ll follow it and push you the log

andynoack, 2017-11-28:
Hi @nocomp,
the dialog you opened at 0:48 in your video has a radio button on the right side that lets you choose between the native backend and a gnuradio backend. Select gnuradio here.

nocomp, 2017-11-28:
what is the exact path of urh.log can t find it
(nothing changed using gnuradio backend)

nocomp, 2017-11-28:
here we go
[WARNING::Device.py::stop_rx_mode] USRP: Receive process is still alive, terminating it
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] USRP: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialogController.py::closeEvent] Device stopped successfully.
[DEBUG::SendRecvDialogController.py::closeEvent] Cleaning up device
[DEBUG::SendRecvDialogController.py::closeEvent] Successfully cleaned up device

nocomp, 2017-11-28:
if ti can help, i never see any lights on on usrp showing wich rx is used, (it has two rx)

nocomp, 2017-11-28:
if you want i have a bit of time this morning we can skype and do some testing

andynoack, 2017-11-28:
Hi @nocomp,
we are busy with a lecture right now but maybe @jopohl can help a little bit in the meantime.

nocomp, 2017-11-28:
my skype id is no-comp if needed

nocomp, 2017-11-28:
new error using gnuradio backend
https://imgur.com/a/RanUL

message in log box
*****************************
UHD Error:
    Device discovery error: EnvironmentError: IOError: Could not load firmware: 
    EnvironmentError: IOError: ihex_reader::read(): record hander returned failure code
Traceback (most recent call last):
  File "D:\Program Files\Universal Radio Hacker\usrp_recv.py", line 104, in <module>
    float(options.bw), str(options.device_args), int(options.port))
  File "D:\Program Files\Universal Radio Hacker\usrp_recv.py", line 51, in __init__
    channels=range(1),
  File "C:\gnuradio\lib\site-packages\gnuradio\uhd\__init__.py", line 122, in constructor_interceptor
    return old_constructor(*args)
  File "C:\gnuradio\lib\site-packages\gnuradio\uhd\uhd_swig.py", line 2629, in make
    return _uhd_swig.usrp_source_make(*args)
RuntimeError: LookupError: KeyError: No devices found for ----->
Empty Device Address
**************************************

message in log file

**************************************
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Init Process
[INFO::AbstractBaseThread.py::initialize_process] Starting Gnuradio
[DEBUG::AbstractBaseThread.py::initialize_process] C:\gnuradio\gr-python27\pythonw.exe D:\Program Files\Universal Radio Hacker\usrp_recv.py --samplerate 1000000.0 --freq 433920000.0 --gain 20 --bandwidth 1000000.0 --port 1337
[INFO::AbstractBaseThread.py::initialize_process] Started Gnuradio
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Process Intialized
[INFO::AbstractBaseThread.py::init_recv_socket] Initalizing receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalized receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Trying to get a connection to gnuradio...
[INFO::AbstractBaseThread.py::init_recv_socket] Got connection
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop
[INFO::AbstractBaseThread.py::stop] Kill grc process
[INFO::AbstractBaseThread.py::stop] Term grc process
[INFO::AbstractBaseThread.py::stop] Could not establish connection to USRP

nocomp, 2017-11-28:
it seems he s trying to use usrp over internet instead of usb, should nt be hard to find 
usrp device can be found with a uhd_find_devices

jopohl, 2017-11-28:
Does it help if you enter type=b200 Into the Device Args?

nocomp, 2017-11-28:
yeaaaaaaaaaaa it worked !!!!!!!
great stuff!!
i ve recorded my window s blinds remote, and replayed the signal, usrp is used on the tx channel propelly, but blinds aren t moving :'( i guess there is a trick somewhere

nocomp, 2017-11-28:
if i may propose a fix, b200 series have two tunner, when you read a signal, URH use rx 2 and when you send signal it use rx/tx 1.
would be more convenient to have only tx/rx 1 used for both action

nocomp, 2017-11-28:
@jopohl can we have a chat on skype?

jopohl, 2017-11-28:
@nocomp I am currently very little on time, but feel free to join [our slack channel](https://join.slack.com/t/stralsundsecurity/shared_invite/enQtMjEwOTIxNzMzODc3LWU4ZWIzMTQ3NDAyNjkzODBhZTJiZDNmN2U0MTA4ZTM1MjhhNTNiYTc4YzQ5MDk2NjU5YzMxOWJmMDQyZDczYjg) where we have lots of discussion related to URH and Software Defined Radios!

closed 2017-11-28 by jopohl