jopohl/urh #370: Bandpass filter error

vsboost, 2017-11-30:
## Steps to Reproduce the Problem

Open signal
Change to Spectrogram view
Select signal
Apply bandpass filter

[image]

Clicking ok continues the process and works.

## Platform Specifications

  - Python Version:
  - Operating System: Windows 10
  - Version of URH: 1.8.15
  - URH was installed: __from source__ /// __via pip__ /// via ```.msi```

jopohl, 2017-11-30:
Thanks for bringing this up. This problem only appears when using the ``` .msi ``` version on windows, so my unittests could not catch it. :wink: 

I just released [version 1.8.16](https://github.com/jopohl/urh/releases/tag/v1.8.16) which adresses this issue.

vsboost, 2017-11-30:
Awesome, thanks again for your hard work.

closed 2017-11-30 by vsboost