jopohl/urh #371: Enable SDRPlay in Windows version

vsboost, 2017-11-30:
I'm unable to enable SDRPlay in windows version .msi

Not sure if it requires a dll file like other sdr's in the C:\Program Files\Universal Radio Hacker directory
Also i have the pothossdr suite installed and am able to use gqrx in windows with the SDRPlay, not sure if that makes a difference or not.
[image]

jopohl, 2017-12-01:
SDRPlay has unfortunately no native support. You can use it via the GNU Radio backend, by defining the directory GNU Radio was installed to at the bottom of the dialog.

vsboost, 2017-12-01:
I tried to, but which directory do you choose? it seems like there needs to be some sort of OK button once the directory is picked, i have tried.
C:\Program Files\PothosSDR\bin
C:\Program Files\PothosSDR\include\gnuradio
C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio

But what ever i choose doesn't seem to stick.

jopohl, 2017-12-01:
It should be the one where GNU Radios ``` python.exe ``` is located at. I am not sure if this is compatible with PothosSDR (never tried that). 

The supported way is to use the official GNU Radio installer from [here](http://www.gcndevelopment.com/gnuradio/downloads.htm).

jopohl, 2017-12-01:
Just had a look at [the source](https://github.com/jopohl/urh/blob/master/src/urh/dev/gr/AbstractBaseThread.py#L53). There should be a directory ``` gr-python27 ``` in the directory you define here. For example, lets assume you installed GNU Radio to ``` C:\Programs ``` then ``` C:\Programs\GNURadio-3.7 ``` will work, because the directory  ``` C:\Programs\GNURadio-3.7\gr-python27 ``` exists.

jopohl, 2017-12-01:
Seems like this woll not work then. Is there a python.exe somewhere in the Pothos Install dir? If so, I could add support for it.

Is it an option for you, to install GNU Radio?

jopohl, 2017-12-01:
What you could also try is to define the path to the python 2 Interpreter (assuming you have one installed). I could imagine, that PothosSDR modifies the PYTHONPATH so it may be enough to just configure a Python 2 interpreter.

vsboost, 2017-12-01:
Unfortunately no python.exe within C:\Program Files\PothosSDR
With the PothosSDR bundle Gnuradio does work and is located in "C:\Program Files\PothosSDR\bin\gnuradio-companion.exe"

My python directory is C:\Python27

vsboost, 2017-12-01:
PYTHONPATH=C:\Program Files\PothosSDR\lib\python2.7\site-packages

jopohl, 2017-12-01:
It may work if you choose the radio button "Python2 interpreter" in the URH dialog from your screenshot and enter ``` C:\Python27 ```.

vsboost, 2017-12-01:
Tried that too, but the settings does not stay. Its like there needs to be an OK button when the directory is chosen if that makes sense.

[image]

vsboost, 2017-12-01:
As soon as you close the dialog box and get back in it defaults back to.
[image]

jopohl, 2017-12-01:
This setting should stay if it works. If there is an error it defaults back to the empty value. You can check if your Python2 interpreter has GNU Radio support with this command:
```
C:\Python27\python.exe -c 'import gnuradio'
```

vsboost, 2017-12-01:
No Go.

C:\Python27>python.exe -c 'import gnuradio'
  File "<string>", line 1
    'import
          ^
SyntaxError: EOL while scanning string literal

C:\Python27>

jopohl, 2017-12-01:
Looks like the Windows CMD behaves a bit weird. If you are in that directory you can also simply enter ``` python.exe ``` this will bring you in the interactive python shell. From here you can then type
```
>>> import gnuradio
```

vsboost, 2017-12-01:
C:\Python27>python.exe
Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:53:40) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio
>>>

jopohl, 2017-12-01:
Is it working when you press enter?

vsboost, 2017-12-01:
Yep, it accepts it when i enter the full filename.
[image]

vsboost, 2017-12-01:
But still cant enable SDRPlay :-(

jopohl, 2017-12-01:
Was the ``` import gnuradio ``` successful after pressing enter in the python interpreter or did it raise an ImportError?

vsboost, 2017-12-01:
No, it just came up blank.

[image]

jopohl, 2017-12-01:
This is actually a good sign! I have a suspicion what is going wrong here. Could you try entering
```
C:\Python27\python.exe -c "import gnuradio"
```
with double quotes instead of single quotes like before and see if this runs without error?

vsboost, 2017-12-01:
Runs without error.

C:\Python27>python.exe -c "import gnuradio"

C:\Python27>

vsboost, 2017-12-01:
Thanks Jopohl,

Your work is truly appreciated.

xykts, 2017-12-04:
windows msi install version  how to solve the problem？  

Thanks jopohl

xykts, 2017-12-04:
sdrplay api is open  ！
Can it be native to support it？

jopohl, 2017-12-04:
Just released [version 1.8.17](https://github.com/jopohl/urh/releases/tag/v1.8.17) which adresses this issue.

@xykts I would like to add a native support. However, we do not possess any devices for testing. If you or someone you know can provide us with two test devices we may add it to our agenda.

xykts, 2017-12-04:
i am windows run  urh(py36)+sdrplay  problems arose

C:\Users\Administrator>urh
Will not regenerate UI, because script can't be found. This is okay in release.
Win32; Microsoft Visual C++ version 14.0; Boost_105900; UHD_003.010.001.001-rele
ase

[DEBUG::SpectrumThread.py::run] Spectrum Thread: Init Process
[INFO::AbstractBaseThread.py::initialize_process] Starting Gnuradio
[DEBUG::AbstractBaseThread.py::initialize_process] C:\Python27\python.exe c:\pyt
hon36\lib\site-packages\urh\dev\gr\scripts\sdrplay_recv.py --samplerate 2000000.
0 --freq 433920000.0 --gain 25 --bandwidth 2000000.0 --port 1337
[INFO::AbstractBaseThread.py::initialize_process] Started Gnuradio
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Process Intialized
[INFO::AbstractBaseThread.py::init_recv_socket] Initalizing receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalized receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Trying to get a connection to gn
uradio...
[INFO::AbstractBaseThread.py::init_recv_socket] Got connection
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop
Traceback (most recent call last):
  File "c:\python36\lib\site-packages\urh\controller\SpectrumDialogController.py
", line 90, in update_view
    if super().update_view():
  File "c:\python36\lib\site-packages\urh\controller\SendRecvDialogController.py
", line 555, in update_view
    new_messages = self.device.read_messages()
  File "c:\python36\lib\site-packages\urh\dev\VirtualDevice.py", line 594, in re
ad_messages
    return self.__dev.read_errors()
  File "c:\python36\lib\site-packages\urh\dev\gr\AbstractBaseThread.py", line 27
1, in read_errors
    return result.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd5 in position 131: invali
d continuation byte

jopohl, 2017-12-04:
Does it work with GNU Radio?

xykts, 2017-12-04:
[image]

it & gnuradio

jopohl, 2017-12-04:
Could you try it with [this](https://ci.appveyor.com/api/buildjobs/v9bnb8no6spdyxbs/artifacts/dist%2FUniversal%20Radio%20Hacker-1.8.17-amd64.msi) installer? I just build it with a version where I catched the Unicode Error.

xykts, 2017-12-04:
i am the pip install version

jopohl, 2017-12-04:
Ah I see, then please replace line 27 ( ``` return result.decode("utf-8") ```) in ``` c:\python36\lib\site-packages\urh\dev\gr\AbstractBaseThread.py ``` with this:
```python
        try:
            return result.decode("utf-8")
        except UnicodeDecodeError:
            return "Could not decode device message"
```

xykts, 2017-12-04:
This is effective. It seems that there is no correct reception

 
[image]

jopohl, 2017-12-04:
Could you share your top_block.py from your GNU Radio example ?

xykts, 2017-12-04:
uplpad sdrplay api , try to build add a native support.  

Thanks jopohl
[SDRplay.zip](https://github.com/jopohl/urh/files/1527801/SDRplay.zip)

xykts, 2017-12-04:
This is my gnuradio top_block.py

[top_block.zip](https://github.com/jopohl/urh/files/1527812/top_block.zip)

jopohl, 2017-12-04:
Does it help if you change ``` c:\pyt
hon36\lib\site-packages\urh\dev\gr\scripts\sdrplay_recv.py :62 ``` from 
```
self.osmosdr_source_0 = osmosdr.source(args="numchan=" + str(1) + " " + "sdrplay")
```
to
```
self.osmosdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + 'sdrplay=0' )
```

Sorry for the long shots, but the SDRplay support was actually added by the community, so I can not verify my ideas because I do not possess an SDRPlay device. This would also make it very hard to develop a native interface.

xykts, 2017-12-04:
Change this" sdrplay=0"

There is no use for anything，In the same way as the above

xykts, 2017-12-04:
Pothos sdr uses soapy device sdrplay 

Soapy imitates UHD devices，I can use USRP as an input
But the received signals are problematic. This may be a problem with soapy

So it's still building native support.

[image]

jopohl, 2017-12-04:
Can you provide us with some test devices or make a PR with native support?

xykts, 2017-12-04:
I'm sorry that my English is poor.
I think I can only do remote testing sdrplay native support
！Sdrplay.com $99 for RSP1A！ I'm not familiar with international shipping
I don't know what I can do

Thank you very much！ I'm here at one o'clock in the morning, and I go to bed first

jopohl, 2017-12-04:
Sure get some sleep mate. :+1: 
I will talk with @andynoack about this. Maybe we have some bucks left so we can buy at least one them, as $99 is not that expensive.

vsboost, 2017-12-05:
Im getting.
[image]

[image]

jopohl, 2017-12-05:
This seems to be an issue with the python2 interpreter. Does it work when you run this in terminal:
```
C:\Python27\python.exe -c "from gnuradio import gr"
```
Note the slight difference to the test we did above.

vsboost, 2017-12-05:
It just returns a blank new line.
[image]

jopohl, 2017-12-05:
What happens if you call:
```
C:\Python27\python.exe "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py"
```
from command line?

vsboost, 2017-12-05:
C:\>cd Python27

C:\Python27>C:\Python27\python.exe "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py"
Traceback (most recent call last):
  File "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py", line 25, in <module>
    from gnuradio import gr
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\__init__.py", line 41, in <module>
    from runtime_swig import *
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
AttributeError: 'module' object has no attribute 'import_module'

C:\Python27>

jopohl, 2017-12-05:
Alright, this is the same error as in the application. Please add ``` print("SYS Path", sys.path) ``` before line 25 of  ``` C:\Program Files\Universal Radio Hacker\sdrplay_recv.py ``` and run it again, so we can see what is in sys.path at this point.

Futhermore, please share the output of ``` C:\Python27\python.exe -c "import sys; print(sys.path)" ``` for comparision.

vsboost, 2017-12-05:
C:\Python27>C:\Python27\python.exe "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py"
('SYS Path', ['C:\\Program Files\\Universal Radio Hacker', 'C:\\Program Files\\PothosSDR\\lib\\python2.7\\site-packages', 'C:\\WINDOWS\\SYSTEM32\\python27.zip', 'C:\\Python27\\DLLs', 'C:\\Python27\\lib', 'C:\\Python27\\lib\\plat-win', 'C:\\Python27\\lib\\lib-tk', 'C:\\Python27', 'C:\\Python27\\lib\\site-packages', 'C:\\Python27\\lib\\site-packages\\gtk-2.0', 'C:\\Python27\\lib\\site-packages\\wx-3.0-msw'])
Traceback (most recent call last):
  File "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py", line 26, in <module>
    from gnuradio import gr
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\__init__.py", line 41, in <module>
    from runtime_swig import *
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 17, in <module>
    _runtime_swig = swig_import_helper()
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\gr\runtime_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)                                                                               AttributeError: 'module' object has no attribute 'import_module'

C:\Python27>
C:\Python27>C:\Python27\python.exe -c "import sys; print(sys.path)"
['', 'C:\\Program Files\\PothosSDR\\lib\\python2.7\\site-packages', 'C:\\WINDOWS\\SYSTEM32\\python27.zip', 'C:\\Python27\\DLLs', 'C:\\Python27\\lib', 'C:\\Python27\\lib\\plat-win', 'C:\\Python27\\lib\\lib-tk', 'C:\\Python27', 'C:\\Python27\\lib\\site-packages', 'C:\\Python27\\lib\\site-packages\\gtk-2.0', 'C:\\Python27\\lib\\site-packages\\wx-3.0-msw']

C:\Python27>

jopohl, 2017-12-05:
I just noticed, that the line numbers in your ``` sdrplay_recv.py ``` do not quite match the latest version. Maybe there went something wrong during install? Please verify, that your ``` C:\Program Files\Universal Radio Hacker\sdrplay_recv.py ``` looks like below. If it doesn't you can safely copy and paste the contents.

```python
#!/usr/bin/env python2
##################################################
# GNU Radio Python Flow Graph
# Title: Top Block
# Generated: Fri Aug 21 15:56:13 2015
##################################################

from optparse import OptionParser
import Initializer

Initializer.init_path()

from gnuradio import gr
from gnuradio.eng_option import eng_option
from grc_gnuradio import blks2 as grc_blks2
from InputHandlerThread import InputHandlerThread
import osmosdr
from gnuradio import zeromq

class top_block(gr.top_block):
    def __init__(self, samp_rate, freq, gain, bw, port):
        gr.top_block.__init__(self, "Top Block")

        ##################################################
        # Variables
        ##################################################
        self.samp_rate = samp_rate
        self.gain = gain
        self.freq = freq
        self.bw = bw

        ##################################################
        # Blocks
        ##################################################
        self.osmosdr_source_0 = osmosdr.source(args="numchan=" + str(1) + " " + "sdrplay")
        self.osmosdr_source_0.set_sample_rate(samp_rate)
        self.osmosdr_source_0.set_center_freq(freq, 0)
        self.osmosdr_source_0.set_freq_corr(0, 0)
        self.osmosdr_source_0.set_dc_offset_mode(0, 0)
        self.osmosdr_source_0.set_iq_balance_mode(0, 0)
        self.osmosdr_source_0.set_gain_mode(False, 0)
        self.osmosdr_source_0.set_gain(gain, 0)
        self.osmosdr_source_0.set_if_gain(gain, 0)
        self.osmosdr_source_0.set_bb_gain(gain, 0)
        self.osmosdr_source_0.set_antenna("", 0)
        self.osmosdr_source_0.set_bandwidth(bw, 0)

        self.zeromq_push_sink_0 = zeromq.push_sink(gr.sizeof_gr_complex, 1, 'tcp://127.0.0.1:' + str(port))

        ##################################################
        # Connections
        ##################################################
        self.connect((self.osmosdr_source_0, 0), (self.zeromq_push_sink_0, 0))

    def get_samp_rate(self):
        return self.samp_rate

    def set_samp_rate(self, samp_rate):
        self.samp_rate = samp_rate
        self.osmosdr_source_0.set_sample_rate(self.samp_rate)

    def get_gain(self):
        return self.gain

    def set_gain(self, gain):
        self.gain = gain
        self.osmosdr_source_0.set_gain(self.gain, 0)
        self.osmosdr_source_0.set_if_gain(self.gain, 0)
        self.osmosdr_source_0.set_bb_gain(self.gain, 0)

    def get_freq(self):
        return self.freq

    def set_freq(self, freq):
        self.freq = freq
        self.osmosdr_source_0.set_center_freq(self.freq, 0)

    def get_bw(self):
        return self.bw

    def set_bw(self, bw):
        self.bw = bw
        self.osmosdr_source_0.set_bandwidth(self.bw, 0)

if __name__ == '__main__':
    parser = OptionParser(option_class=eng_option, usage="%prog: [options]")
    parser.add_option("-s", "--samplerate", dest="samplerate", help="Sample Rate", default=100000)
    parser.add_option("-f", "--freq", dest="freq", help="Frequency", default=433000)
    parser.add_option("-g", "--gain", dest="gain", help="Gain", default=30)
    parser.add_option("-b", "--bandwidth", dest="bw", help="Bandwidth", default=200000)
    parser.add_option("-p", "--port", dest="port", help="Port", default=1337)
    (options, args) = parser.parse_args()
    tb = top_block(float(options.samplerate), float(options.freq), int(options.gain),
                   float(options.bw), int(options.port))
    iht = InputHandlerThread(tb)
    iht.start()
    tb.start()
    tb.wait()
```

If it then should complain about a missing Initalizer.py file, just create that file in that directory with this conent:
```python
import sys

import os
import tempfile

def init_path():
    # Append script path at end to prevent conflicts in case of frozen interpreter
    sys.path.append(sys.path.pop(0))

    try:
        with open(os.path.join(tempfile.gettempdir(), "gnuradio_path.txt"), "r") as f:
            gnuradio_path = f.read().strip()

        os.environ["PATH"] = os.path.join(gnuradio_path, "bin") + os.pathsep + os.environ["PATH"]
        sys.path.insert(0, os.path.join(gnuradio_path, "lib", "site-packages"))

    except IOError:
        pass
```

However, a clean reinstall (with prior uninstall) may be the better option here.

vsboost, 2017-12-05:
The file was different, did an uninstall and clean install,

Traceback (most recent call last):
  File "C:\Program Files\Universal Radio Hacker\sdrplay_recv.py", line 18, in <module>
    from gnuradio import zeromq
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\zeromq\__init__.py", line 35, in <module>
    from probe_manager import probe_manager
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\zeromq\probe_manager.py", line 22, in <module>
    import zmq
ImportError: Bad magic number in C:\Program Files\Universal Radio Hacker\zmq\__init__.pyc

vsboost, 2017-12-05:
Also noticed a crash when pressing the X to close the window.
[image]

jopohl, 2017-12-05:
What does 
```
C:\Python27\python.exe -c "from gnuradio import zeromq"
```
give you?

vsboost, 2017-12-05:
C:\Python27>python.exe -c "from gnuradio import zeromq"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\zeromq\__init__.py", line 35, in <module>
    from probe_manager import probe_manager
  File "C:\Program Files\PothosSDR\lib\python2.7\site-packages\gnuradio\zeromq\probe_manager.py", line 22, in <module>
    import zmq
ImportError: No module named zmq

C:\Python27>

jopohl, 2017-12-05:
Seems like you have no zeromq installed for you python2 interpreter. You could try this:
```
C:\Python27\python.exe -m pip install pyzmq
```

vsboost, 2017-12-05:
Getting close now lol

[image]

gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.11.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf airspy soapy redpitaya 
[1m[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.6/LMS7Support.dll)
  LoadLibrary() failed: The specified procedure could not be found.

[0m

FATAL: 
GR-UHD detected ABI compatibility mismatch with UHD library.
GR-UHD was build against ABI: 3.10.2,
but UHD library reports ABI: 3.10.1
Suggestion: install an ABI compatible version of UHD,
or rebuild GR-UHD component against this ABI version.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

vsboost, 2017-12-05:
This may be an issue with Pothos by the looks of it, so don't worry too much about it.

jopohl, 2017-12-05:
Looks also like a PothosSDR error for me, maybe Pothos is not supposed to be used in this way.

You still have the option to install the [official GNU Radio](http://www.gcndevelopment.com/gnuradio/downloads.htm) for Windows, which is better supported. If you do so, you need to configure the GNU Radio install dir in URH options.

Unfortunately, we do not have the budget at the moment to afford a SDRPlay. I would add native support otherwise.

vsboost, 2017-12-05:
Thanks Johannes for trying, ill try to reach out to the SDRPlay team and see what they can arrange, do you have an email address i could send them to contact you?

jopohl, 2017-12-05:
Sure, just use my GMail address: Johannes.Pohl90@gmail.com
Thanks for your support.

[1 more reply]

closed 2017-12-06 by jopohl