jopohl/urh #382: Autoassign labels

ikarus23, 2017-12-22:
Hi! Is there a way to assign labels automatically? Did I miss it? I see the "Assign labels" checkbox in the in the dropdown menu next to the "Analyze" button...

Lets say I have a demodulated stream like this:
```
0001111010111111101010000 | 0101010101010101 | 000000000000 | 01001011011100010100010111011000100010 | 00000 | 11111111111 | 0000000 | 00011001000010100000
                                   1                                                2                               3                          4
```
I know "1" is the preamble, "2" is the address, "3" the payload and "4" the crc. The rest is irrelevant trash (with fixed length). Is there a way to auto assign the label "preamble" to the bit sequence of 1 and then specify the other labels with a relative offset? Removing the "trash" should be easy once everything is labelled, because I can use the "Show only labels in protocol" feature of URH.

jopohl, 2017-12-29:
Short answer: This is not yet supported. It is coming 2018.
Long answer: Currently, the auto assign feature tries to automatically assign labels based on heuristics. We are working on improving this mechanism behind the scenes. Ideally, it won't be even necessary to specify labels manually because URH finds them itself.

For your current task you may find the ZeroHide Plugin useful, to hide all the zero trash and assign labels afterwards. This assumes, of course, that the trash always consists of zeros.

ikarus23, 2017-12-29:
Thanks for the feedback and thanks again for this great tool. Looking forward to this feature ;)

closed 2019-05-20 by jopohl

ikarus23, 2019-05-20:
Thank you. This project has come a far way! If I could star it again, I would! ;)

PR #403 (2019-05-20): More accurate logic analysis
This PR aims to greatly enhance the accuracy of automatic label assignment and find labels for Preamble, Sync, Addresses, Sequence Number and Checksum. This will fix #78  and fix #382.