jopohl/urh #388: Delete from Analysis Tab

muhantu, 2018-01-11:
In URH, we can delete signal from interpretation tab, but can we delete just from analysis tab (per cell(s))?

In analysis tab, we only can hide rows, no delete per row or column..

jopohl, 2018-01-12:
You have two ways to do that:
1. Select the bits you want to delete -> right click -> ``` Show selection in Interpretation ```. Here you can delete the corresponding part of the original signal. This is the preferred way to delete trash.
2. Right click anywhere on Analysis table and click on context menu ``` Writeable (decouples from signal) ```. Then you can delete bits using the Delete key inside the table.

muhantu, 2018-01-12:
1. I know that, and usually use that..
2. I have try that, but that's mean interpretation and analysis is separate?

muhantu, 2018-01-12:
Ok, I think `Writeable` solve my problem.. I can check and uncheck that..
Thanks @jopohl :D

jopohl, 2018-01-12:
You're welcome. :)

closed 2018-01-12 by jopohl