jopohl/urh #389: RTLSDR-OPEN: Error Code: -12 (-12)

kaba2211, 2018-01-15:
Hi,

I got an USB RTL2832 + R820T2 device, which I wanted to use in URH. The USB driver were changed to WinUSB 6.1.xxx and the device is working well in SDR#.
But when I try to use it in URH Spectrum analyser, there is an "RTLSDR-OPEN: Error Code: -12 (-12)" error message. 
Is it a problem in URH or a driver issue? What does error -12 refer to?

Thanks!

jopohl, 2018-01-15:
Not sure what the "-12" refers to. The error code comes from the rtl_sdr driver itself. Just did a quick grep on rtl_sdr's source code but could not find anything useful.

Did the stick work with URH before you updated the USB driver?

Another thing that comes to my mind is that the stick could be hold by another application e.g. SDR# so maybe try unplug it, close Spectrum analyzer dialog and replug the stick again.

alfredgamulo, 2018-01-16:
@kaba2211 I got this issue a couple nights ago as well. 
I found the solution here: https://www.rtl-sdr.com/forum/viewtopic.php?f=7&t=1797

Basically, you have to install the driver not for `"Bulk Interface (0)` but instead after choosing `"List All Devices"` pick the `Composite` device.

I hope that works!

kaba2211, 2018-01-16:
Hi Alfred,
I just tried to change the driver as described in the Link, and Ta-da! It works well.
Thank you for your help.

jopohl, 2018-01-16:
Thanks for sharing your solution @alfredgamulo . I added a message box with a link to this issue in #391 in case RTL-SDR error code -12 appears on Windows, so other users will be directed here.

closed 2018-01-16 by jopohl

dwarning, 2021-05-19:
Sorry for reopen this issue.
I am working under ubuntu 20.04 and have installed URH from snap store, Version 2.9.2.
If I activate the spectrum analyzer or the recorder I got:
RTLSDR-OPEN (#0): Error Code: -1 (-1)
RTLSDR: failed to start rx mode
BTW: same error if I not connect the stick into USB port.
All my other rtl-sdr tools are working: GQRX, Gnu Radio or rtl_433.
blacklist dvb_usb_rtl28xxu is there and rtl-sdr.rules is in /etc/udev/rules.d

n6151h, 2021-06-10:
Seeing same problem on ubuntu 20.04 VM (VirtualBox) hosted on windows 10.
gqrx, rtl_433, etc all working.

davidwkerr, 2022-05-09:
I have the same problem Ubuntu 22.04, 20.04 (both native) and 19.4 as a VM. gqrx and other apps are fine.