jopohl/urh #393: Analog Devices Adalm Pluto support?

peterg10, 2018-01-23:
The device support in urh is excellent but is there any chance of support for the Analog Devices Adalm Pluto. The specs are here: https://wiki.analog.com/university/tools/pluto but the highlights are RX/TX 325-3800MHz (easily modded to 6000MHz) 12bit ADC and DAC, 20MHz instantaneous bandwidth.

jopohl, 2018-01-23:
Sounds cool! I would love to add support, but I see two obstacles for it:

1. I could not find an API after a quick google search. However, on the link you provided they say that they are currently rewriting the docs so this may be just a matter of time.
2. More imporantly, I do not possess any Pluto SDRs for testing. Maybe you know someone that could provide me with one or two test devices?

peterg10, 2018-01-23:
AFAIK (I'm no developer) the API/driver this through **libiio** : [https://github.com/analogdevicesinc/libiio](url)

The second point that could be a problem. AD haven't been able to ship them due to a shortage of the DDR RAM chip they used. They've changed the BOM to use a larger memory chip so they should be available soon.

herfemp, 2018-02-28:
Interested indeed,
I just got a unit from digikey.ca today(even if they were showing 165 on back order). I would be available for any kind of testing and provide feedback !

jopohl, 2018-02-28:
Thanks for your offer @herfemp but I find it difficult to develop hardware support in such a remote manner.
I see, however, there is some interest here. I just reached out to AD if they can provide me with some test devices.

ivanenev, 2018-03-02:
It would be grate if you can support this devise . One can see that all shops are working with per-orders. There is a increasing interest in this devise. I got one  from Arrow Electronic last week.

reald, 2018-04-05:
Implemented some basic PlutoSDR support. Used gnuradio for this, so _no_ native support. Receiving seems to work quite well but transmitting with generator only produces garbage. (Data comes to slow out of urh?) So quality is not good enough for a pull request but maybe it is helpful anyway. If you want to try transmitting gain=0 seems to work best.

See fork at https://github.com/reald/urh

DG9BFC, 2018-07-17:
if you add adalm pluto ... please also add the lime board

vsboost, 2018-07-17:
LimeSDR is supported. https://github.com/jopohl/urh/wiki/Supported-devices

theMadCatter, 2018-11-17:
I would loan you my Pluto if that would help.

jopohl, 2018-11-17:
Thanks @theMadCatter but we just bought some Plutos a week ago. I am actually already working on a support, guess it will be ready in a week or so.

DG9BFC, 2018-11-19:
thanks jopohl .... so now lime and pluto are supported ... that is very cool cause these are two very cheap but good rx AND tx sdr setups ....
i have an XTRX on order ... that also will be a nice sdr setup an hopefully also supported by URH?!?!?

jopohl, 2018-11-19:
Thanks @DG9BFC . XTRX might be supported when we can get our hands on such a thing. :wink:

peterg10, 2018-11-19:
My thanks too for implementing this.

I had to use: `MACOSX_DEVELOPMENT_TARGET=10.14 pip3 install --upgrade urh` to get it to build on Mojave. Otherwise I got an error that it couldn't  `include complex`

DG9BFC, 2018-11-20:
jopohl ... you should hurry up and order one of the xtrx cs version (only 260 bucks) cause soon there will be only the pro version available (more then double the price)
https://www.crowdsupply.com/fairwaves/xtrx/updates/announcing-xtrx-pro
the pro version has a bigger fpga chip and a better master clock ... but thats all ... so i am guessing that the better buy is the cs version (and soon only the pro version will be available and in my view it is tooo costly)
on the software side both versions should run the same software
you shoud hurry up cause there may be just a handful units left (6th of november is 2 weeks ago so maybe all 250 units are already sold out?!?)
greetz
sigi

PR #574 (2018-11-19): Add native support for PlutoSDR
This PR adds native support for Adalm PlutoSDR and fix #393.

### Todo
- [x] integrate RX
- [x] integrate TX
- [x] windows drivers
- [x] windows test
- [x] revert appveyor
- [x] update changelog
- [x] update wiki