jopohl/urh #399: Exception: Could not find python 2 interpreter. Make sure you have a running gnuradio installation.

Mavrikant, 2018-02-28:
## Expected Behavior
I have installed GNURadio 3.7.11, GQRX and libraries for RTL-SDR, HackRF.  GR and GQRX work perfectly. I have no idea what is wrong. 

## Actual 

> serdar@Mavrikant-Ubuntu:~$ urh
Will not regenerate UI, because script can't be found. This is okay in release.
[INFO::SettingsProxy.py::get_receive_buffer_size] Try to allocate receive buffer with size 2,139GB
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/urh/dev/gr/ReceiverThread.py", line 30, in run
    self.initialize_process()
  File "/usr/local/lib/python3.6/dist-packages/urh/dev/gr/AbstractBaseThread.py", line 211, in initialize_process
    raise Exception("Could not find python 2 interpreter. Make sure you have a running gnuradio installation.")
Exception: Could not find python 2 interpreter. Make sure you have a running gnuradio installation.

## Steps to Reproduce the Problem

  1. Install URH via pip3
  2. Record signal from RLT-SDR or HackRF
  3. Nothing happens

## Platform Specifications

  - Python Version: Python 3.6.4+,  Python 2.7.14+
  - Operating System: Ubuntu 18.04, Linux 4.15.0-10-generic, 64 bit
  - Version of URH: 2.0.0 
  - URH was installed: sudo pip3 install urh

jopohl, 2018-02-28:
Since you are using RTL-SDR / HackRF you could use the native Backend for these devices ( Options -> Device) which is the recommended backend anyway.

Mavrikant, 2018-02-28:
Native backend was gray
I installed `librtlsdr-dev` and` libhackrf-dev`.
Hit the rebuild.
Restart URH.
Native backend is still gray. 
But recoding **works** with GNURadio. 

Terminal output:
> serdar@Mavrikant-Ubuntu:~$ urh
Will not regenerate UI, because script can't be found. This is okay in release.
[INFO::SettingsProxy.py::get_receive_buffer_size] Try to allocate receive buffer with size 2,660GB
[INFO::AbstractBaseThread.py::initialize_process] Starting Gnuradio
[DEBUG::AbstractBaseThread.py::initialize_process] /usr/bin/python2 /usr/local/lib/python3.6/dist-packages/urh/dev/gr/scripts/rtl-sdr_recv.py --samplerate 1000000.0 --freq 433920000.0 --gain 25 --bandwidth 1000000.0 --port 1337 --freq-correction 1 --direct-sampling 0
[INFO::AbstractBaseThread.py::initialize_process] Started Gnuradio
[INFO::ReceiverThread.py::run] Initialize receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalizing receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalized receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Trying to get a connection to gnuradio...
[INFO::AbstractBaseThread.py::init_recv_socket] Got connection
[INFO::AbstractBaseThread.py::stop] Kill grc process
[INFO::AbstractBaseThread.py::stop] Term grc process
[INFO::AbstractBaseThread.py::stop] Stopped receiving: Stop button clicked

Recoding logs:
> gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!

Is this makes sense for you? It works for me but I would like to help you.

jopohl, 2018-02-28:
Is it still gray after a reinstallation of URH? (pip uninstall urh // pip install urh)?

If nothing helps you would simply need to configure the path to your python2 interpreter in Options -> Device so it can use the python2 interpreter for GNU Radio backend.

Mavrikant, 2018-02-28:
About the previous message, it was working with **GNURadio backend** after installing librtlsdr-dev and libhackrf-dev.

I reinstalled URH like you said. Native backend support works now. Thanks for your help.

The interesting thing is lack of native backend packages effects GNURadio backend. Because I don't remember changing anything.

closed 2018-02-28 by jopohl

phretor, 2018-03-17:
@jopohl still, if one needs to use the GNURadio backend (e.g., for the BladeRF), the aforementioned exception is thrown when python2 is not found.

BTW, it's worth keeping an eye on this: https://github.com/roseengineering/osmocat

jopohl, 2018-03-17:
@phretor I am unable to reproduce this Error so for better debugging please tell me:

- What is your OS?
- What does python2 --version give you?