jopohl/urh #416: Can't install URH 2.0.1 with pip3

jJit0, 2018-03-23:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior

## Actual Behavior
`Collecting urh
  Using cached urh-2.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/user/0/pip-build-ndu0xeek/urh/setup.py", line 19, in <module>
        from src.urh.dev.native import ExtensionHelper
    ModuleNotFoundError: No module named 'src.urh'
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/user/0/pip-build-ndu0xeek/urh/ `

## Steps to Reproduce the Problem

  1. pip3 install urh
  OR
  2. python3 setup.py install

## Platform Specifications

  - Python Version: 3.6.5
  - Operating System: Kali 2018.1
  - Version of URH: 2.0.1
  - URH was installed: __from source__ /// __via pip__ /// via ```.msi```
    Tried from pip3 and from git clone

jopohl, 2018-03-24:
Can't reproduce this (just verified with official Kali docker image). Maybe there are some leftovers from previous installations? Please try:
```
sudo pip3 uninstall urh
pip3 uninstall urh
sudo updatedb
locate urh
```
verfiy nothing is left and then install again with ``` sudo pip3 install urh ```

jJit0, 2018-03-24:
I reproduce your steps:
1- rm /root/.config/urh
2- rm /usr/local/lib/python3.5/dist-packages/urh*
3- pip3 install -U urh
`File "<string>", line 1, in <module>
      File "/tmp/user/0/pip-build-330x02e_/urh/setup.py", line 19, in <module>
        from src.urh.dev.native import ExtensionHelper
    ModuleNotFoundError: No module named 'src.urh'
`

jopohl, 2018-03-24:
Just tried it on a fresh Kali Linux VM ``` pip3 install -U urh ``` worked without any problems. I also cloned the source and ran ``` python3 setup.py install ``` which worked, too. Not sure what could be the issue here, sorry.

It might be worth to give a +1 in the tool request [here](https://bugs.kali.org/view.php?id=4211) to get URH in the Kali repos and avoid such problems in the future.

jopohl, 2018-03-26:
closing here. drop a comment if you think this should be reopened.

closed 2018-03-26 by jopohl