jopohl/urh #418: LimeSDR MIni?

Ken98045, 2018-03-29:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
It should work with the LimeSDR Mini

## Actual Behavior
Whenever I try the SA or to record the mini status LEDs go from blinking green to red , or if they are red to begin with they don't go green like in other apps such as SDR Console.  I don't think that this is the expected behavior.  Nothing gets recorded. Using the SA I get nothing or garbage.

## Steps to Reproduce the Problem

  1.Start any functions
  2.No valid response, nothing gets recorded.
  3.

## Platform Specifications

  - Python Version:
  - Operating System:Windows 10 Pro/Core I7
  - Version of URH: 2.01
  - URH was installed: __from source__ /// __via pip__ /// via ```.msi```
msi

andynoack, 2018-03-30:
Hey, we currently do not support the LimeSDR Mini. We do not have any of these devices here for testing, is anyone in contact with Lime Microsystems?

vsboost, 2018-07-07:
My LimeSDR-Mini works both SA and record.

Using the Windows version of URH 2.2.1

[image]

jopohl, 2018-07-07:
Nice, I updated the bundled Lime driver a while ago, seems like supported was added with that version. Thanks for verifying!

closed 2018-07-07 by jopohl