jopohl/urh #419: How to import/use URH from script

gaardiolor, 2018-03-31:
Hello! I was trying to write some simple python code to demodulate FSK, but I keep getting one or two bitflips with my code, it's not good enough. URH is however, when I 

- Open the complex file
- Noise: 0.01
- Center: 0.3425
- Bit length: 31
- Error tolerance: 5
- Modulation: FSK

I'm getting exactly the bits I should get, it works perfectly! But, how can I get those bits in my script ? I'm opening the complex file like this:

```
def read_complex_array(filename):
    samples = numpy.fromfile(filename, dtype=numpy.complex64)
    signal_samples = []
    for sample in samples:
        signal_samples.append((sample.real, sample.imag))
    return signal_samples
```

Which modules should I import, how can I set the parameters and execute the demodulation ? If you can point me in the good direction that would be great, thanks!

jopohl, 2018-03-31:
As a starting point you may have a look at ``` urh/tests/test_demodulations.py ```. Everything you ask for should be covered there. Let me know if you have more specific questions.

gaardiolor, 2018-03-31:
Exactly the help I needed!

```
from urh.signalprocessing.ProtocolAnalyzer import ProtocolAnalyzer
from urh.signalprocessing.Signal import Signal

def main():
    signal = Signal('/path/to/file', "FSK-Test")
    signal.modulation_type = 1
    signal.bit_len = 31
    signal.qad_center = 0.3425
    signal.noise_threshold = 0.01
   
    proto_analyzer = ProtocolAnalyzer(signal)
    proto_analyzer.get_protocol_from_signal()
    print(proto_analyzer.plain_bits_str[0])

if __name__ == "__main__":
    main()
```

Thanks!

closed 2018-03-31 by gaardiolor