jopohl/urh #420: Feature Request: Byte order

cyko69, 2018-04-05:
It would help if I could choose the byte order for every message field.

jopohl, 2018-04-05:
Could you explain your use case a bit more?

You can already 
  1. choose the byte order in label view to change it for certain labels at the bottom of analysis
   [image]
  2. choose "Change Bitorder" in decoding to apply it for all messages.
   [image]

cyko69, 2018-04-16:
Was on vaccation, so sorry for the late answer.
Both options are changing the BIT order, i need to have a different BYTE order (endianess).
e.g. I got a 0111110000000001 (0x7c 0x01) and i need to interpret this as ‭000101111100‬ (0x01 0x7c),
so the decimal value i want to have is (dec) 380 (and not 31745).

jopohl, 2018-04-16:
Ah thanks for the clarification. So we are talking about a new Column "endianess" next to "Bit order" in the label table at the bottom?

PR #428 (2018-04-17): make endianness selectable in order column
fix #420
closed 2018-04-17 by jopohl