jopohl/urh #431: Homebrew test block fails for 2.0.2

ilovezfs, 2018-04-22:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
```
  test do
    xy = Language::Python.major_minor_version "python3"
    ENV.prepend_create_path "PYTHONPATH", libexec/"lib/python#{xy}/site-packages"
    (testpath/"test.py").write <<~EOS
      from urh.util.GenericCRC import GenericCRC;
      c = GenericCRC();
      expected = [1, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 1, 1]
      assert(expected == c.crc([1, 2, 3]).tolist())
    EOS
    system "python3", "test.py"
  end
```
should pass

## Actual Behavior
The test block fails.

```
bash-3.2$ python3
Python 3.6.5 (default, Mar 30 2018, 06:41:49) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.42.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from urh.util.GenericCRC import GenericCRC;
>>> c = GenericCRC();
>>> expected = [1, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 1, 1]
>>> assert(expected == c.crc([1, 2, 3]).tolist())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AssertionError
>>> c.crc([1, 2, 3]).tolist()
[1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 1, 1]
>>> 
````

Note that `[1, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 1, 1]` != `[1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 1, 1]`

## Steps to Reproduce the Problem

  1. install urh with numpy-1.14.2, numpy-1.14.2, pyzmq-17.0.0
  2. run `brew test urh`

## Platform Specifications

  - Python Version: 3.6.5
  - Operating System: OS X 10.11
  - Version of URH: 2.0.2
  - URH was installed: source build with Homebrew (trying to upgrade Homebrew formula to 2.0.2)

ilovezfs, 2018-04-22:
Here you can see it failing on all 3 of our build bots (macOS 10.11, 10.12 and 10.13):
https://github.com/Homebrew/homebrew-core/pull/26954

jopohl, 2018-04-22:
I have some questions one this:

1. What is the reason that you defined this test instead of using the unit tests that come with URH?
2. Did this test work with 2.0.1? I just did a git diff and could not find a relavant change for the CRC between 2.0.1 and 2.0.2.
3. How did the ``` expected ``` variable evolve? It seems a bit odd to me, given that the CRC function should only accept arrays with 1s and 0s (bits). We also have some tests for the CRC [here](https://github.com/jopohl/urh/blob/master/tests/test_CRC.py) which work as expected. Nevertheless, there may a bug here. @andynoack may tell more about this, I am just trying to understand how you came to the ``` expected ``` value.

ilovezfs, 2018-04-22:
>What is the reason that you defined this test instead of using the unit tests that come with URH?

The test block is an integration test that makes sure the software works after it's already installed not during the build.

>Did this test work with 2.0.1? I just did a git diff and could not find a relavant change for the CRC between 2.0.1 and 2.0.2.

It passes with 2.0.0 and fails with 2.0.1 and 2.0.2.

> How did the expected variable evolve? It seems a bit odd to me, given that the CRC function should only accept arrays with 1s and 0s (bits). We also have some tests for the CRC here which work as [...]

I'll look.

ilovezfs, 2018-04-22:
It was originally
```
  test do
    xy = Language::Python.major_minor_version "python3"
    ENV.prepend_create_path "PYTHONPATH", libexec/"lib/python#{xy}/site-packages"
    system "python3", "-c", "import urh.util.crc;c = urh.util.crc.crc_generic();assert(c.crc([1, 2, 3]) == [True, False, False, False, False, False, True, True, False, False, False, False, False, False, True, True])"
  end
```
in https://github.com/Homebrew/homebrew-core/commit/125d2528b59b77638bac4b066a0f98c7460fa266

And I changed it to the above in https://github.com/Homebrew/homebrew-core/commit/b8a28535798bc5c386dac9d590b6fa08b9381436

andynoack, 2018-04-22:
We have changed the padding algorithm (already before 2.0.1) for CRCs. Earlier there was a padding to multiple of 8 Bits, that is removed now. This may explain the different expected values, because now it is possible to calculate CRCs over 3 bits like in your test (note that 2 and 3 are no valid Bit values).

ilovezfs, 2018-04-22:
```
diff --git a/Formula/urh.rb b/Formula/urh.rb
index 3d50de9..82bcb15 100644
--- a/Formula/urh.rb
+++ b/Formula/urh.rb
@@ -60,7 +60,7 @@ class Urh < Formula
       from urh.util.GenericCRC import GenericCRC;
       c = GenericCRC();
       expected = [1, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 1, 1]
-      assert(expected == c.crc([1, 2, 3]).tolist())
+      assert(expected == c.crc([1, 2, 3, 0, 0, 0, 0, 0]).tolist())
     EOS
     system "python3", "test.py"
   end
```
and it passes.

andynoack, 2018-04-22:
Great! Ist there a special reason to test non binary values?

andynoack, 2018-04-22:
What about [0,1,0,1,1,0,1,0] or something like that?