jopohl/urh #435: urh-2.0.4 build error: src/urh/dev/native/lib/hackrf.cpp:607:10: fatal error: config.h: No such file or directory

yarda, 2018-05-07:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
No build error

## Actual Behavior
build error:
...
src/urh/dev/native/lib/hackrf.cpp:607:10: fatal error: config.h: No such file or directory

## Steps to Reproduce the Problem

  1. Build for Fedora rawhide (29) (https://koji.fedoraproject.org/koji/taskinfo?taskID=26833486)
  2.
  3.

## Platform Specifications

  - Python Version: python-3.6.5-4
  - Operating System: Fedora rawhide (29)
  - Version of URH: 2.0.4
  - URH was installed: __from source__ 

I tried to build with hackrf-2017.02.1 and hackrf-2018.01.1. Unfortunately, there is no config.h, nor the HACKRF_HAS_MULTI_DEVICE defined.

yarda, 2018-05-07:
Same problem with the RTLSDR_HAS_BANDWIDTH for rtl-sdr.

jopohl, 2018-05-07:
That's strange...I just had a look in the [latest source release](https://github.com/jopohl/urh/archive/v2.0.4.tar.gz)  and the ``` config.h ``` is located in ``` src/urh/dev/native/includes/ ```. I also just installed the 2.0.4 version of URH on arch linux and did not experience any issues.

Where do you download the sources from? Maybe you need to tell Fedora something about the new include path (``` src/urh/dev/native/includes/  ```)?

yarda, 2018-05-07:
Hmm, it seems it isn't in the .tar.gz:
https://github.com/jopohl/urh/archive/v2.0.4.tar.gz

But it is in the zip.

yarda, 2018-05-07:
Hmm, I re-downloaded the tar.gz and it's there. Maybe the download was corrupted.

yarda, 2018-05-07:
Ah, sorry, got it, it seems to be fault in our build procedure, sorry for noise.

closed 2018-05-07 by yarda