jopohl/urh #438: Checksum calculation wrong

sycophantic, 2018-05-22:
## Expected Behavior
Calculate correct checksum using polynomial 31.
a1a502cb0819080003d0020000 should have checksum 28. See [CRC8/MAXIM](http://crccalc.com/?crc=a1%20a5%2002%20cb%2008%2019%2008%2000%2003%20d0%2002%2000%2000&method=crc8&datatype=hex).

## Actual Behavior
Checksum calculation is wrong.
Checksum calculated is 6a, see screenshot.

[image]

## Platform Specifications

  - Python Version: 3.6.5
  - Operating System: Ubuntu 18.04
  - Version of URH: 2.0.4
  - URH was installed: via pip

andynoack, 2018-05-22:
Actually the computation of the CRC is not wrong but you need two parameters (RefIn=true and RefOut=true -> input and output bits are inverted/reflected) to calculate it in your way. We are about to implement these, stay tuned :)

closed 2018-05-22 by jopohl

sycophantic, 2018-05-22:
Great. Thanks!

PR #439 (2018-05-22): make refin and refout configurable for CRC
Add options to reflect input (RefIn) and output (RefOut) for CRC caluclations.
fix #438