jopohl/urh #443: Feature request: Select bit in "Interpretation" and jump to them in "analysis"

Nesos-ita, 2018-05-29:
Im late, i'll polish this later...
Related to, read this before continuing: https://github.com/jopohl/urh/issues/442  
Since i now have a singe loooong line of bits and i had to sacrifice the super useful pause marker to obtain this (pause threshold=0 + zerohide enabled) i have no easy way to set the labels  
for example there is a sync at the beginning but to know how long is i have to re-enable the pause marker add a label, and disable it again.

jopohl, 2018-05-29:
Did you try to set the pause threshold to 16 instead of 0? As the pause between actual messages is much greater than between sync and data I see no reason to disable message breaking entirely with pause threshold 0.

Nesos-ita, 2018-05-29:
i could and it will probably work but then i will have 4 messages (instead of 7):  
 - preamble  
 - sync + data (repeated three times)  

The problem is that i will have preamble and sync+data and there is no point in comparing them, also i could not set a label as preamble and other as sync+data because they will be on the same line (again split message problem).   
i could also cut away the preamble or any part except data but this means editing N signals and is also problematic if i have to replay it after.  
If the signal is made in that way i would like to keep every part of it, eventually mark the first part as preamble and add it's related label but not deleting it.

jopohl, 2018-05-29:
Ah I see...well at least this will get the preambles into seperate messages so you could hide them in Analysis.

It it also possible to deal with the labeling problem using __message types__. Here is an example:
[image]

Message 2 has a different message type than the other messages so it can be labeled differently. You can add a new message type using the + button and assign a message type to a message using the context menu in Analysis table. You can even automate this assigning by creating a rule for the message type using the button left to the "+". This way you could, e.g, assign a "preamble" message type to all preamble messages.

Btw thanks for taking your time to make such detailed issues, it really helps fixing bugs quickly. :+1:

Nesos-ita, 2018-05-29:
> Btw thanks for taking your time to make such detailed issues, it really helps fixing bugs quickly. +1  
Happy to be useful ^_^  

i'll try message type this evening, now i'm not at home