jopohl/urh #451: Wrong hiding when using multiple message types and labels

Nesos-ita, 2018-05-29:
## Expected Behavior  
I have two message types, one label in the first, and two labels in the second.  
Hiding the label in the first message type should hide only the labelled bits, and only of that message type.  

## Actual Behavior  
If a label say "bit 1-31" bits 1-31 are hidden regardless of the message type

## Steps to Reproduce the Problem

 1. Add two message types and one label per message type  
 2. set two rows as two different message types  
 3. hide one label  
 4. (part of) the other is hidden too  

Check the photo:  
[image]

## Platform Specifications

  - Python Version: installed 3.6 but it uses 3.5 included with the msi
  - Operating System: win8.1 x64
  - Version of URH: 2.0.4 MSI installer x64
  - URH was installed: via msi

Nesos-ita, 2018-05-29:
mmm so how do i solve my problem? i'd like to see only "message" and not preamble/sync, possibly without editing all the signals to keep only the data part.  
also what is the point of having multiple message types if i can't keep enabled only one? for example if i have a transerceiver and i set two message types "query" and "answer" and i'm interested only in answers  
i'm a bit confused

jopohl, 2018-05-29:
Well, in your case it would be much easier to use a Filter located at the top of Analysis as shown below:
[image]

You would simply enter, e.g., the synchronization bits, hit "Filter" and all messages not containing this pattern will be hidden, which are the preambles in your case. Note, you must first select the "Filter" as the button uses "Search" as default.

The problem with the bit hiding is not that easy to solve. If you simply leave bits out e.g. by replacing them with a "-" you loose the ability of alignment. However, I see your point, it may confuse users as it is not the expected behaviour.

Nesos-ita, 2018-05-29:
Nice, it works :)  i'm still a bit confused for the hiding, i have to think more about it.  
the only left problem is that hidden labels are not kept hidden:  
 - if i click mark diff (enable and disable), on disable hidden labels are shown again  
 - if i click filter (maybe this one is wanted since i'm searching/filtering for this)

Nesos-ita, 2018-05-29:
seems that it happens if:  
 - you have two message types  
 - i setted one label per message type  
 - one label is a column other is a row and they *must* collide/intersecate  
 - hidden a label  
 - enabled and disabled show diff  

same happens if i enable and disable the first signal (but not the second)

jopohl, 2018-05-30:
Thanks I was able to reproduce this behaviour and fixed it in #452.

closed 2018-05-30 by jopohl