jopohl/urh #454: Question: Generating unmodulated signals (no carrier). Am i doing it in the correct way?

Nesos-ita, 2018-06-04:
Hi, i have recorded a signal with the soundcard by tapping the data line going to the rf transmitter, so i recorded the square wave unmodulated. 
i was trying to use generator to replicate this so that i can do replay attack directly in the data line or by connecting an rf trasmitter that receive it's data from the soundcard.  
aka the poor's man hack rf... only one frequency (443,92MHz) but still it transmit software defined data :)  
Thus i need to generate unmodulated signal.  
I have done it with some success:  
[image]  
Is this the right way to do it? is there any other way?  
for me right now the only problem is that i soldered only the left channel of the jack that goes to the breadbord so i have to export to audacity, separate stereo tracks in two mono, delete the left muted channel and keep the right playied on the left channel.  
Not a big issue... i can solder the other wire or swap it.  
Other thing, in the edit window and send window the displaied signal voltage is the opposite:  
[image]  
"edit window" that goes 0 to lower matches theone viewed in audacity. i guess that the "send window" is different due to the fact that is complex signal???

andynoack, 2018-06-05:
Hey @Nesos-ita, I think it is more effective to clear up such questions in direct dialog. Hence feel free to join #urh in our Slack-Channel (there is an invite link on our main page) that is dedicated to discussions!

Nesos-ita, 2018-06-05:
is there a slack channel? nice to see! i'm coming ^_^

closed 2018-06-06 by andynoack