jopohl/urh #455: Question: is it possible to decode this in your sw? (similar to differential manchester)

Nesos-ita, 2018-06-05:
I don't like the idea of asking questions here, as it is a place to post issues, but you can always close this as "not issue, ask to google/duckduckgo"  
I have this signal where:
 - 001 means 1 (because of long signal+ignore/skip the next)  
 - 01 means 0 (because is short+skip next)  
is similar to differential manchester, in fact: no transition means 1, tansition means 0.  
bit if i use that it throw many errors because there is never 11, only 00.  
[image]  
if it is not possible i'll do it by sw, and maybe also the generation.

jopohl, 2018-06-05:
Did you try to create a new decoding with a substiution table like this?
[image]

andynoack, 2018-06-05:
Maybe you are looking for a custom decoding like "Invert, Morse Code (1,2,1)"

Nesos-ita, 2018-06-05:
Substitution doesn't work, it's also in the description "use only same length".  
but invert+morse code is exactly what i was looking for!  
i applied that decoding, all the labels, and this program it's just AWESOME!!!

andynoack, 2018-06-05:
You can only have one decoding per line. For special purposes, however, you can exclude arbitrary labels from applying that selected encoding of the line (Do Not Decode flag).

closed 2018-06-06 by Nesos-ita