jopohl/urh #457: Simulator doesn't send anything using soundcard

Nesos-ita, 2018-06-05:
My problem: sending arbitrary data that i write decoded and is encoded by the proggram  
My solution: use the simulator  

## Expected Behavior
I have read the userguide pdf and seems that the simulator should send real data (like the generator).  
i have decoded a protocol, add all the labels, added a recorded signal to the simulator, edited the data (from the menu down i selected a different temperature and humidity)  
when i click start simulation i should hear sound from the soundcard.  
also:  
 - simulation-->messages-->open in analysis button doesn't work  
 - setting a field as "live input" instead of "constant" should probably ask me to input something but it doesn't.  

## Actual Behavior
i clicked simulate but soundcard is mute (while using generator i can hear sound).  

i'll add simulation log:  
Status:
```
giu 5 20:27:20.733579: Waiting for devices
giu 5 20:27:21.734824: Waiting for devices
giu 5 20:27:22.667441: TX is ready to operate
giu 5 20:27:22.735486: Simulation is running
giu 5 20:27:22.735486: Sending message 1
preamble: 11001100110011001100110011001100110011001100110011001100110011
giu 5 20:27:22.736486: Sending message 2
synchronization: 1010101010101010101
giu 5 20:27:22.737487: Sending message 3
temp: 012
temp2: 3
hum: 45
giu 5 20:27:22.737487: Sending message 4
synchronization: 1010101010101010101
giu 5 20:27:22.738488: Sending message 5
temp: 012
temp2: 3
hum: 45
giu 5 20:27:22.738488: Sending message 6
synchronization: 1010101010101010101
giu 5 20:27:22.739488: Sending message 7
temp: 012
temp2: 3
hum: 45
giu 5 20:27:22.739488: Stop simulation: Finished
```

Messages:  
```
1 (T->Broadcast): 11001100110011001100110011001100110011001100110011001100110011
2 (T->Broadcast): 1010101010101010101
3 (T->Broadcast): 000000010010001101000101
4 (T->Broadcast): 1010101010101010101
5 (T->Broadcast): 000000010010001101000101
6 (T->Broadcast): 1010101010101010101
7 (T->Broadcast): 000000010010001101000101
```
Devices:  
```
SoundCard: Initializing pyaudio...
SoundCard: Initialized pyaudio
SoundCard: Successfully started pyaudio stream
SoundCard: successfully started tx mode
SoundCard-CLOSE: Success
```
## Platform Specifications

  - Python Version: installed 3.6 but it uses 3.5 included with the msi
  - Operating System: win8.1 x64
  - Version of URH: 2.0.4 MSI installer x64
  - URH was installed: via msi

jopohl, 2018-06-05:
Please note that the table in Generator is __editable__ so you can enter arbritrary bits here. It seems that you do not need to fire up the Simulator as it is intended to be used for stateful attacks only.

Using the right click menu you can also duplicate lines in the Generator table or insert additional columns. If you then send out the data, URH will take care of applying the encoding for you.

jopohl, 2018-06-05:
The same holds for the constant values in the simulator: They are editable after double clicking, so there is no need to pop up a dialog. :wink: 
The live input ones are learned during simulation time, so it would make no sense to enter a value here.

I will close this issue. Let me know if you feel it should be reopened.

closed 2018-06-05 by jopohl

Nesos-ita, 2018-06-06:
please clarify this: **should simulator tranmit data to the soundcard for real?** from the userguide seems yes. **if yes there is a bug** as if i use it with the soundcard nothing is transmitted (generator works, simulator not).  

secondary things: i know that generator fields are editable, i used simulator because it has this nice way to input data (anyway generator has the viewtype hex, which is similar)  
[image]