jopohl/urh #464: urh.desktop isn't installed

yurivict, 2018-06-18:


jopohl, 2018-06-18:
Please give more details:
-What OS are you using?
- How did you install URH? (pip / from source / package)?
- What version of URH are you referring to?

yurivict, 2018-06-18:
OS: FreeBSD
How did I install: using distutils
Version is 2.1.1.

jopohl, 2018-06-18:
The .desktop file does not get installed via distutils, it is only used when you install URH using a package manager on a supported distro (Arch, Fedora, Gentoo, Suse).

However, the icons should be there. Can you check that you are using the bundled icons and not the native ones? (Edit -> Options -> View -> Choose icon theme)

yurivict, 2018-06-18:
Arch installs the icon and the desktop files manually: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=urh

You should install them through ```setup.py```, to do things the same way on all platforms.

jopohl, 2018-06-18:
Ah you are talking about ``` appicon.png ``` and not the whole icon set! 

I do not do this on purpose, since I cannot guarantee that __all__ linux flavors behave in the same way Arch does. Furthermore, you may then need to ``` sudo ``` the installation just because of the desktop icon.

yurivict, 2018-06-18:
> you may then need to sudo the installation just because of the desktop icon.

No, desktop file isn't any different compared to other installed files in terms of sudo. Everything is normally installed under the DESTDIR first.

> I cannot guarantee that all linux flavors behave in the same way Arch does.

For example, qTox, which installs on all linuxes with cmake, installs these files into the same location everywhere except for apple: https://github.com/qTox/qTox/blob/master/cmake/Installation.cmake#L39 It probably also works on Windows.

Your situation shouldn't be any different, ```urh``` would be better if you do the same.

jopohl, 2018-06-18:
The situation with ``` setup.py ``` is a bit more complicated than with cmake when it comes to deploying such data files, as you can see e.g. [in this thread](https://stackoverflow.com/questions/501597/how-to-distribute-desktop-files-and-icons-for-a-python-package-in-gnome-with). However, the mentioned workaround there has a big issue with itself, since dependency management with PyPi would not work anymore (see [here](https://mail.python.org/pipermail/distutils-sig/2011-December/018218.html)).

jopohl, 2018-06-19:
I will close this one, as there seems to be a kind of convention to not do this task with setuptools:
> For the installation of files which are not application data files (like images, UI files etc) but provide integration into the operating system, you are better off with using a real packaging format (like RPM or deb). - https://stackoverflow.com/a/501624

Desktop file for URH gets already installed in the following cases:
- You are on Windows and use the msi installer
- You are on OSX and use ``` brew install urh ```
- You are on Arch Linux / Fedora / Suse / Gentoo and use your package manager to install URH

So the problem is, that there is no FreeBSD package for URH, maybe someone will create one in the future? :wink: 
 
Otherwise, if you can point me to a Python based project that deploys .desktop files with setuptools (I did not found any) in a reliable manner I might reconsider this decision. Just drop a comment here then and I will reopen.

closed 2018-06-19 by jopohl