jopohl/urh #474: simulator, soundcard, doesn't send any audio.

Nesos-ita, 2018-06-23:
## Expected Behavior
Related to: https://github.com/jopohl/urh/issues/457  
Using simulator and soundcard i should hear something as it happens in the generator.  

## Actual Behavior
Nothing heared, well one time it worked but i can't reproduce it.  
i clicked start simulation so many times but only *once* i heard some sound, in no way i can hear anything again.  

## Steps to Reproduce the Problem  
Unclear how to reproduce, for me it never works: carrierFrequency=0, bit len=11, modulation: ask, sample rate: 44,1k, amplitude for 0=0% amplitude for 1 = 100%, phase 270°; only sending is enabled, no receiving.  

since it wasn't working, I have downloaded the latest "SoundCard.py" source file and placed it in "lib\urh\dev\native", deleted the pyc file. verieifed that it is used by placing a division by 0. it throw exception when i click start simulation so it works, removed the division, added import ctypes.  
i have added a messagebox at the beginning of each function:  
ctypes.windll.user32.MessageBoxW(0, "function init_device called", "msg", 0)

  1. click simulate, a new window opens
  2. click start

note that under the "send_sync" function i'have placed this:  
ctypes.windll.user32.MessageBoxW(0, "sending: {}".format(data), "msg", 0)  

poor's man execution trace:  
init_device  
setup_device  
prepare_sync_send  
pack_complex  
send_sync (sending: [0.])  
prepare_sync_send  
pack_complex  
send_sync (sending: <multiprocessing.sharedctypes.c_float_Array_2048 object at 0x...>)  
shutdown_device  

This for the **First time** i click start.  
because if i click start again: prepare_sync_send is executed again but no pack_complex and no send_sync.  
if i close and reopen the window, the execution trace is again the first shown.  
not always i see "sending [0.]" followed by "sending: <multipro.." sometimes i see [0.] two times...  

## Platform Specifications

  - Python Version: 3.6 x64, but programuse it's own 3.5
  - Operating System: win 8.1 x64
  - Version of URH: v2.1.1
  - URH was installed: via msi x64

jopohl, 2018-06-23:
I just did a little experiment on this:

1. Go to generator, right click -> add empty message
2. set modulation to FSK
3. set pause to 0
4. Play that (__one sending repeat__) -> Can hear a sound. I also exported the signal using the  "Generate file" as a ``` .wav ``` and played it via VLC which produced the same sound

Now I changed the bit length of the modulation to __10__. I could __not__ hear a sound in the generator with one sending repeat. I also exported the file as  ``` .wav ``` again to play it with VLC which also remained silent.

So I think the problem here is the short bit length and the resulting short sample sequences. I suspect the sound card needs a minimum of data to turn on or the generated sound is too short for the human ear. I am not an audio expert though, but as VLC also does not play a sound I am quite confident that this is not a bug in URH.

Nesos-ita, 2018-06-24:
> I suspect the sound card needs a minimum of data to turn on  

i had the same idea in fact i tried to add 1000 zero bits at the beginning and at the end to make a long audio.  
still without result.  

> sound is too short for the human ear  

i don't think that this is the problem as sending exactly the same data using the generator i can hear it perfectly.  

Also, i tried again the egenerator and i see 9 messagebox "sending: ..." and each time i press ok i hear a piece of the transmission (except for one which is full pause and the last one, probably again pause).  
The messageboxes looks like this: "sending: [1.924881e-08 -1.0000000e+00 1.1924881e-08 ... 0.0000000e+00 0.0000000e+00 0.0000000e+00]"

doing the same in the simulator i see only 1 messagebox: "sending: [0.]"  
so i think that the data is not sent.  
other interesting thing: if i wait too much (few seconds) before pressing ok on "pack_complex" it say "simulation finished" and prints all the 0 and1 even before i clicked ok and there is no call ever to "send_sync"  
maybe there is a second thread killed after a too short timeout?  
i have set repeat=1, maximum retries=1, timeout=100000ms

jopohl, 2018-06-24:
> generator has a bug: the last part of the transmission is not sent immediatly but it is sent after a looong pause (check the recordings)

Are you referring to the pause at the end? This is configurable by double clicking at the pause, shown below:
[image]

Nesos-ita, 2018-06-24:
Here you can see the exported signal and the generated one; same settings, a part is missing and generated only after an unwanted delay.  
[image]  
The program added in the screen that you sent all the appropriate pauses in the signal.  Plus one *after* the end of the signal.  
the program selectedvalue for the pause is: 367 samples (8,32ms)  
setting it to anything below 700 samples (15,87ms) causes that delay.  
it doesn't matter if the pause is long 2 samples or 600 samples, still there is that delay and it is always at the same distance

jopohl, 2018-06-24:
Yeah I just added a padding for that in f119353f938b5588d7a66f05064673ebf091d8ba . Now Generator produces always a sound regardless how short the signal is.

Nesos-ita, 2018-06-24:
i have tested it locally and it fixes the generator problem.
but simulator still doesn't work.
(i'm online on slack if you need to ask something, connect with team viewer, whatever...)

jopohl, 2018-06-24:
Simulator issue should be fixed with d6f18cb69965e76cf824af979bd0ac4857d22e6d.

PR #476 (2018-06-24): pad short sounds when sending data
fix #474
closed 2018-06-24 by jopohl