jopohl/urh #481: installation failed: Cython.Compiler.Errors.InternalError: Internal compiler error: 'config.pxi' not found

blshkv, 2018-06-30:
```
>>> Creating Manifest for /data/pentoo/pentoo/net-wireless/urh
>>> Existing ${T}/environment for 'urh-2.2.0' will be sourced. Run 'clean'
>>> to start with a fresh environment.
>>> Checking urh-2.2.0.tar.gz's mtime...
>>> WORKDIR is up-to-date, keeping...
>>> It appears that 'pretend' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.pretended' to force pretend.
>>> It appears that 'setup' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.setuped' to force setup.
>>> It appears that 'unpack' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.unpacked' to force unpack.
>>> It appears that 'prepare' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.prepared' to force prepare.
>>> It appears that 'configure' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.configured' to force configure.
>>> It appears that 'compile' has already executed for 'urh-2.2.0'; skipping.
>>> Remove '/var/tmp/portage/net-wireless/urh-2.2.0/.compiled' to force compile.
>>> Test phase [not enabled]: net-wireless/urh-2.2.0

>>> Install urh-2.2.0 into /var/tmp/portage/net-wireless/urh-2.2.0/image/ category net-wireless
 * python3_5: running distutils-r1_run_phase distutils-r1_python_install
python3.5 setup.py --with-hackrf --without-rtlsdr --without-usrp --without-airspy --without-limesdr install --root=/var/tmp/portage/net-wireless/urh-2.2.0/image/_python3.5
Skipping native usrp support
Skipping native airspy support
Enforcing native hackrf support
Skipping native limesdr support
Skipping native rtlsdr support
Skipping native support for sdrplay
Traceback (most recent call last):
  File "setup.py", line 147, in <module>
    ext_modules=get_extensions(),
  File "setup.py", line 102, in get_extensions
    extensions = cythonize(extensions, compiler_directives=COMPILER_DIRECTIVES, quiet=True)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 818, in cythonize
    aliases=aliases)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 719, in create_extension_list
    kwds = deps.distutils_info(file, aliases, base).values
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 590, in distutils_info
    return (self.transitive_merge(filename, self.distutils_info0, DistutilsInfo.merge)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 600, in transitive_merge
    node, extract, merge, seen, {}, self.cimported_files)[0]
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 612, in transitive_merge_helper
    sub_deps, sub_loop = self.transitive_merge_helper(next, extract, merge, seen, stack, outgoing)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 605, in transitive_merge_helper
    deps = extract(node)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 581, in distutils_info0
    externs = self.cimports_and_externs(filename)[1]
  File "/usr/lib64/python3.5/site-packages/Cython/Utils.py", line 44, in wrapper
    res = cache[args] = f(self, *args)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 491, in cimports_and_externs
    for include in self.included_files(filename):
  File "/usr/lib64/python3.5/site-packages/Cython/Utils.py", line 44, in wrapper
    res = cache[args] = f(self, *args)
  File "/usr/lib64/python3.5/site-packages/Cython/Build/Dependencies.py", line 474, in included_files
    include_path = self.context.find_include_file(include, None)
  File "/usr/lib64/python3.5/site-packages/Cython/Compiler/Main.py", line 274, in find_include_file
    error(pos, "'%s' not found" % filename)
  File "/usr/lib64/python3.5/site-packages/Cython/Compiler/Errors.py", line 177, in error
    raise InternalError(message)
Cython.Compiler.Errors.InternalError: Internal compiler error: 'config.pxi' not found
 * ERROR: net-wireless/urh-2.2.0::pentoo failed (install phase):
 *   (no error message)
 * 
 * Call stack:
 *     ebuild.sh, line  124:  Called src_install
 *   environment, line 3654:  Called distutils-r1_src_install
 *   environment, line  969:  Called _distutils-r1_run_foreach_impl 'distutils-r1_python_install'
 *   environment, line  400:  Called python_foreach_impl 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 3199:  Called multibuild_foreach_variant '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 2368:  Called _multibuild_run '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line 2366:  Called _python_multibuild_wrapper 'distutils-r1_run_phase' 'distutils-r1_python_install'
 *   environment, line  680:  Called distutils-r1_run_phase 'distutils-r1_python_install'
 *   environment, line  937:  Called distutils-r1_python_install
 *   environment, line  853:  Called esetup.py 'install' '--root=/var/tmp/portage/net-wireless/urh-2.2.0/image/_python3.5'
 *   environment, line 1497:  Called die
 * The specific snippet of code:
 *       "${@}" || die "${die_args[@]}";
 * 
 * If you need support, post the output of `emerge --info '=net-wireless/urh-2.2.0::pentoo'`,
 * the complete build log and the output of `emerge -pqv '=net-wireless/urh-2.2.0::pentoo'`.
 * The complete build log is located at '/var/tmp/portage/net-wireless/urh-2.2.0/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/net-wireless/urh-2.2.0/temp/environment'.
 * Working directory: '/var/tmp/portage/net-wireless/urh-2.2.0/work/urh-2.2.0'
 * S: '/var/tmp/portage/net-wireless/urh-2.2.0/work/urh-2.2.0'

```

The previous version (2.1.0) was installing fine. Any idea what's wrong with 2.2.0?

jopohl, 2018-06-30:
Just released version 2.2.1 which should fix this issue.

closed 2018-07-01 by blshkv

vineet3101, 2024-12-29:
`    Getting below error while running on pycharm using python. Unable to run the program due to below error:

    ERROR: Failed building wheel for TA-Lib-Precompiled
    ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (TA-Lib-Precompiled)
    > Package :TA-Lib-Precompiled                                                                  
    > Python Version : Python 3.12.1
    > Cython version 3.0.11
    
    Please help in finding the solution !!
    Below are the logs :
    
    > Collecting TA-Lib-Precompiled
      Using cached TA-Lib-Precompiled-0.4.25.tar.gz (276 kB)
      Preparing metadata (setup.py): started
      Preparing metadata (setup.py): finished with status 'done'
    Requirement already satisfied: numpy in c:\python312\lib\site-packages (from TA-Lib-Precompiled) (1.26.4)
    Building wheels for collected packages: TA-Lib-Precompiled
      Building wheel for TA-Lib-Precompiled (setup.py): started
      Building wheel for TA-Lib-Precompiled (setup.py): finished with status 'error'
      Running setup.py clean for TA-Lib-Precompiled
    Failed to build TA-Lib-Precompiled
    
    >WARNING: Ignoring invalid distribution ~ (C:\Python312\Lib\site-packages)
    WARNING: Ignoring invalid distribution ~-p (C:\Python312\Lib\site-packages)
    WARNING: Ignoring invalid distribution ~ip (C:\Python312\Lib\site-packages)
    WARNING: Ignoring invalid distribution ~~p (C:\Python312\Lib\site-packages)
      error: subprocess-exited-with-error
      
      >python setup.py bdist_wheel did not run successfully.
      exit code: 1
      
      >[102 lines of output]
      C:\Python312\Lib\site-packages\setuptools\__init__.py:94: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
      !!
      
              ********************************************************************************
              Requirements should be satisfied by a PEP 517 installer.
              If you are using pip, you can try `pip install --use-pep517`.
              ********************************************************************************
      
    >  !!
        dist.fetch_build_eggs(dist.setup_requires)
      running bdist_wheel
      running build
      running build_py
      creating build\lib.win-amd64-cpython-312\talib
      copying talib\abstract.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\deprecated.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\stream.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_abstract.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_data.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_func.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_pandas.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_polars.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\test_stream.py -> build\lib.win-amd64-cpython-312\talib
      copying talib\__init__.py -> build\lib.win-amd64-cpython-312\talib
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\Jai\AppData\Local\Temp\pip-install-9i7a4_y6\ta-lib-precompiled_354bc0160a70451b8e32e1b0d92cb391\setup.py", line 146, in <module>
          setup(
        File "C:\Python312\Lib\site-packages\setuptools\__init__.py", line 117, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\setuptools\_distutils\core.py", line 183, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\setuptools\_distutils\core.py", line 199, in run_commands
          dist.run_commands()
[... 46 lines cut ...]
          deps = extract(node)
                 ^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\Cython\Build\Dependencies.py", line 687, in distutils_info0
          cimports, externs, incdirs = self.cimports_externs_incdirs(filename)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "Cython\\Utils.py", line 129, in Cython.Utils.cached_method.wrapper
        File "C:\Python312\Lib\site-packages\Cython\Build\Dependencies.py", line 574, in cimports_externs_incdirs
          for include in self.included_files(filename):
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "Cython\\Utils.py", line 129, in Cython.Utils.cached_method.wrapper
        File "C:\Python312\Lib\site-packages\Cython\Build\Dependencies.py", line 556, in included_files
          include_path = self.context.find_include_file(include, source_file_path=filename)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\Cython\Compiler\Main.py", line 299, in find_include_file
          error(pos, "'%s' not found" % filename)
        File "C:\Python312\Lib\site-packages\Cython\Compiler\Errors.py", line 178, in error
          raise InternalError(message)
      Cython.Compiler.Errors.InternalError: Internal compiler error: '_common.pxi' not found
      [end of output]
      
    >  note: This error originates from a subprocess, and is likely not a problem with pip.
      ERROR: Failed building wheel for TA-Lib-Precompiled
    ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (TA-Lib-Precompiled)

`

blshkv, 2025-01-01:
This bug is fixed and closed. Please file a new one with your issue.