jopohl/urh #486: UserWarning: Unknown distribution option: 'long_description_content_type'

yurivict, 2018-07-02:
Distutils prints this warning with urh version 2.2.1:
```
/usr/local/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
```

The FreeBSD port.

jopohl, 2018-07-02:
This warning should go away if you upgrade setuptools to the most recent version ``` python3 -m pip install --upgrade setuptools ```.

jopohl, 2018-07-04:
Alternatively you could install it with ``` pip3 install . ``` instead of ``` python3 setup.py install ```

closed 2018-07-04 by jopohl

harishahamed26, 2024-03-04:
/usr/local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:265: UserWarning: Unknown distribution option: 'long_description_content'
        warnings.warn(msg)
      /usr/local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:265: UserWarning: Unknown distribution option: 'project_url'
        warnings.warn(msg)
        
     Got the above error.
     I did upgrade as well and the version is  "**setuptools in /usr/local/lib/python3.11/site-packages (69.1.1)**"