jopohl/urh #487: Feather Request (?): Save received to data hex file.

anthropic-radio-source, 2018-07-02:
Hi all

Feature request (?)

I am fairly new to URH but think it is a marvelous tool. I have looked but cannot find any easy way to save the bit/hex/ascii data to a separate file for analysis by other tools. Currently, I cut/past the data from either the Interpretation or Analysis widow and save to a text file. Unfortunately, this is saved as in ASCII and need a bit of furtling to get it into a hex file.

Is there any way I can save it straight to a hex format file?

Regards

ARS

anthropic-radio-source, 2018-07-02:
Sorry title should read Feature Request (?): Save received data to data hex file.

Spell checker was too quick for me.

jopohl, 2018-07-02:
Thanks for your request, this is indeed not possible right now, but should be quite easy to add. We will have a look at it!

anthropic-radio-source, 2018-07-02:
Thanks. It will save me loads of time.

PR #488 (2018-07-03): allow save and load of binary protocols
This PR aims to fix #487 . Protocols can now be saved in binary format (``` .bin ``` from Analysis via the save button:
[image]

Vice versa, loading a ``` .bin ``` protocol is possible via the load button right next to the save button or using File -> Open.
closed 2018-07-03 by jopohl