jopohl/urh #490: Feature request: Please provide an AppImage for Linux

probonopd, 2018-07-03:
URH is a very useful application but can be a bit of a hassle to get running on Linux depending on which distribution you are using, which Python version you have, how you install Python packages etc. Currently it is definitely more work than "download one file, set the executable bit, run it".

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

__Ultimaker Cura__ is an example of a very complex Python application being distributed as an AppImage.

https://github.com/AppImage/AppImageKit/wiki/Bundling-Python-apps should get you started. If you have questions, AppImage developers are on #AppImage on irc.freenode.net.

jopohl, 2018-07-04:
Currently URH is in the repositories of Arch / Suse / Fedora and Gentoo. For other distributions the installation is as simple as ``` pip3 install urh ```. If you want to have all dependencies preinstalled you can use the URH [docker image](https://hub.docker.com/r/jopohl/urh/).

Of course, you need to have docker installed for this, which is not the case for an AppImage. However, I see different problems that would arise here: For example, if I install hackrf via ubuntu package there are automatically udev rules installed so I do not have to be root to access the USB device. On the other hand, even if we bundle hackrf driver in the AppImage we could not install udev rules via AppImage, so we would have to be root to run it when we want to access USB devices or install udev rules afterwards.

So if anyone wants to put work on this, I will accept a PR for it, but we have currently no plans to dive into this.

probonopd, 2018-07-04:
We can install udev rules via AppImage. For Arduino, I am using this as part of a custom `AppRun`/start script which appears to be working for me, at least on Live ISOs (more testing on installed systems needed):

```
cat > /tmp/roothelper <<\EOoF                                                                                                                                                           
#!/bin/bash                                                                                                                                                                             
cat > /etc/udev/rules.d/50-arduino.rules <<\EOF                                                                                                                                         
SUBSYSTEM=="tty", KERNEL=="ttyUSB[0-9]*|ttyACM[0-9]*", GROUP="users", MODE="0666"                                                                                                       
EOF                                                                                                                                                                                     
udevadm trigger
EOoF
chmod a+x /tmp/roothelper

sudo true && sudo /tmp/roothelper
sudo true || pkexec --disable-internal-agent /tmp/roothelper || true

rm /tmp/roothelper
```

Intended behavior:
* On Live ISOs, it just becomes root, does its thing, without asking the user
* On installed systems, it asks the user for the root password