jopohl/urh #502: Cannot install urh on Raspbian Stretch

NicHub, 2018-07-14:
I cannot install `urh` on my Raspberry Pi 2 with a fresh install of Raspbian Stretch. When I run the command `sudo pip3 install urh`, I get the following error:

	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting urh
	  Using cached https://www.piwheels.org/simple/urh/urh-2.2.2-cp35-cp35m-linux_armv7l.whl
	Collecting pyqt5 (from urh)
	  Could not find a version that satisfies the requirement pyqt5 (from urh) (from versions: )
	No matching distribution found for pyqt5 (from urh)

Note that I installed these libraries before:

	sudo apt-get --assume-yes install python3-numpy python3-psutil python3-zmq python3-pyqt5 g++ libpython3-dev python3-pip cython3

	sudo apt-get --assume-yes install qt5-default pyqt5-dev pyqt5-dev-tools

Any idea how to install `urh` on a Raspberry?

jopohl, 2018-07-14:
I just installed URH on a Pi 3 the other day, so generally speaking it is possible. :wink:

Please post the output of the following commands:

```
python3 --version
python3 -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)"
sudo pip3 install pyqt5
```

NicHub, 2018-07-14:
Thank you for your answer. Here are the outputs:

	😺  13:01:59 – pi@raspberrypi:~ > python3 --version
	Python 3.5.3

	😺  13:03:08 – pi@raspberrypi:~ > python3 -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)"
	5.7.1

	😺  13:03:15 – pi@raspberrypi:~ > sudo pip3 install pyqt5
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting pyqt5
	Could not find a version that satisfies the requirement pyqt5 (from versions: )
	No matching distribution found for pyqt5

jopohl, 2018-07-14:
Looking good so far, maybe your pip3 is too old? You can check with
```
pip3 --version
```
If you get anything below 10 you can update it with
```
pip3 install --upgrade pip
```

If it is not working after that please try using pip like this:
```
python3 -m pip install --upgrade pip
python3 -m pip install pyqt5
```

NicHub, 2018-07-14:
Here are the outputs from the commands you proposed:

	😺  13:09:17 – pi@raspberrypi:~ > pip3 --version
	pip 10.0.1 from /usr/local/lib/python3.5/dist-packages/pip (python 3.5)

	😺  13:22:03 – pi@raspberrypi:~ > pip3 install --upgrade pip
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Requirement already up-to-date: pip in /usr/local/lib/python3.5/dist-packages (10.0.1)

	😺  13:22:20 – pi@raspberrypi:~ > pip3 --version
	pip 10.0.1 from /usr/local/lib/python3.5/dist-packages/pip (python 3.5)

	😺  13:22:31 – pi@raspberrypi:~ > python3 -m pip install --upgrade pip
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Requirement already up-to-date: pip in /usr/local/lib/python3.5/dist-packages (10.0.1)

	😺  13:22:56 – pi@raspberrypi:~ > python3 -m pip install pyqt5
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting pyqt5
	  Could not find a version that satisfies the requirement pyqt5 (from versions: )
	No matching distribution found for pyqt5

	😺  13:23:10 – pi@raspberrypi:~ > python3 -m pip install --upgrade pip
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Requirement already up-to-date: pip in /usr/local/lib/python3.5/dist-packages (10.0.1)

	😺  13:26:17 – pi@raspberrypi:~ > python3 -m pip install pyqt5
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting pyqt5
	  Could not find a version that satisfies the requirement pyqt5 (from versions: )
	No matching distribution found for pyqt5

	😺  13:26:41 – pi@raspberrypi:~ > pip3 install --upgrade pip
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Requirement already up-to-date: pip in /usr/local/lib/python3.5/dist-packages (10.0.1)

	😺  13:27:50 – pi@raspberrypi:~ > python3 -m pip install pyqt5
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting pyqt5
	  Could not find a version that satisfies the requirement pyqt5 (from versions: )
	No matching distribution found for pyqt5

jopohl, 2018-07-14:
Can you install it like this:
```
git clone https://github.com/jopohl/urh
cd urh
pip3 install .
```

The very strange thing here is, that pip3 wants to install pyqt5 although it is already installed on your system.

NicHub, 2018-07-14:
It worked, thank you very much!

> Note that I had to use `sudo`:

	git clone https://github.com/jopohl/urh
	cd urh
	sudo pip3 install .

jopohl, 2018-07-14:
You're welcome!

closed 2018-07-14 by jopohl