jopohl/urh #509: zmq.error.Again: Resource temporarily unavailable

boundmania, 2018-07-21:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
normal operation

## Actual Behavior
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/urh/dev/gr/SpectrumThread.py", line 32, in run
    rcvd += recv(32768)  # Receive Buffer = 32768 Byte
  File "zmq/backend/cython/socket.pyx", line 792, in zmq.backend.cython.socket.Socket.recv
  File "zmq/backend/cython/socket.pyx", line 828, in zmq.backend.cython.socket.Socket.recv
  File "zmq/backend/cython/socket.pyx", line 193, in zmq.backend.cython.socket._recv_copy
  File "zmq/backend/cython/socket.pyx", line 188, in zmq.backend.cython.socket._recv_copy
  File "zmq/backend/cython/checkrc.pxd", line 19, in zmq.backend.cython.checkrc._check_rc
zmq.error.Again: Resource temporarily unavailable
Abort trap: 6

## Steps to Reproduce the Problem

  1. urh start
  2. select spectrum analyzer
  3. start

## Platform Specifications

  - Python Version:
Python 2.7.14
Python 3.7.0
  - Operating System: mac os 10.13.4
  - Version of URH: 2.2.2
  - URH was installed: via pip3

jopohl, 2018-07-21:
What SDR are you using? Easiest way might be to switch to native backend. If this is not possible: Does a ``` python3 -m pip install --upgrade pyzmq ``` help?

jopohl, 2018-07-24:
Closing here due to no response. Just drop a comment if you need it reopened.

closed 2018-07-24 by jopohl

uhaba, 2018-08-07:
control@ctp:~$ urh
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Init Process
[INFO::AbstractBaseThread.py::initialize_process] Starting Gnuradio
[DEBUG::AbstractBaseThread.py::initialize_process] /usr/bin/python2 /usr/local/lib/python3.6/dist-packages/urh/dev/gr/scripts/rtl-sdr_recv.py --samplerate 1000000.0 --freq 433000000.0 --gain 25 --bandwidth 1000000.0 --port 1337 --freq-correction 1 --direct-sampling 0
[INFO::AbstractBaseThread.py::initialize_process] Started Gnuradio
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Process Intialized
[INFO::AbstractBaseThread.py::init_recv_socket] Initalizing receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalized receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Trying to get a connection to gnuradio...
[INFO::AbstractBaseThread.py::init_recv_socket] Got connection
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/urh/dev/gr/SpectrumThread.py", line 32, in run
    rcvd += recv(32768)  # Receive Buffer = 32768 Byte
  File "zmq/backend/cython/socket.pyx", line 792, in zmq.backend.cython.socket.Socket.recv
  File "zmq/backend/cython/socket.pyx", line 828, in zmq.backend.cython.socket.Socket.recv
  File "zmq/backend/cython/socket.pyx", line 193, in zmq.backend.cython.socket._recv_copy
  File "zmq/backend/cython/socket.pyx", line 188, in zmq.backend.cython.socket._recv_copy
  File "zmq/backend/cython/checkrc.pxd", line 19, in zmq.backend.cython.checkrc._check_rc
zmq.error.Again: Resource temporarily unavailable

jopohl, 2018-08-07:
Please install a recent version of RTL-SDR, go to URH -> Options and rebuild native extensions, restart URH and after that select native backend for RTL-SDR.

uhaba, 2018-08-07:
When running this I get
Skipping native support for airspy
Skipping native support for hackrf
Skipping native support for limesdr
Skipping native support for rtlsdr
Skipping native support for usrp
Skipping native support for sdrplay
running build_ext

I get this with the test command:
control@ctp:~$ rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.
Reattached kernel driver

This is a bit repetitive but I see these in the terminal after running rebuild:
control@ctp:~$ urh
Skipping native support for airspy
Skipping native support for hackrf
Skipping native support for limesdr
Skipping native support for rtlsdr
Skipping native support for usrp
Skipping native support for sdrplay
Skipping native support for airspy
Skipping native support for hackrf
Skipping native support for limesdr
Skipping native support for rtlsdr
Skipping native support for usrp
Skipping native support for sdrplay
running clean
'build/lib' does not exist -- can't clean it
'build/bdist.linux-x86_64' does not exist -- can't clean it
'build/scripts-3.6' does not exist -- can't clean it

Checking drivers, they are up to date as well:
control@ctp:~$ sudo apt-get install rtl-sdr
[sudo] password for control: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
rtl-sdr is already the newest version (0.5.4-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

When clicking this button without really any other changes other than setting the radio, it crashes to desktop with the previous error.
[image]

I appreciate the help. I will share with the class if you have any more ideas.

jopohl, 2018-08-07:
You are missing the ``` -dev ``` package of RTL-SDR (see [here](https://github.com/jopohl/urh#generic-way-with-pip-ubuntudebian)), so a simple ``` sudo apt-get install librtlsdr-dev ``` should enable you to build the native extension with URH.

uhaba, 2018-08-07:
Awesome! Thanks for the quick response! That resolved it. I will share out.