jopohl/urh #517: Recording via URH creates mirror image of signal

vsboost, 2018-07-29:
I have noticed this before but never reported it, what i find is recording a signal from URH seems to create a mirror of the signal not too far off the original, the picture below is of an ASK signal from a weather sensor.

[image]

Here is the same signal but recorded using SDR-Console
[image]

This is using the sdrplay, haven't tested with other sdr's - i notice the same behavior when using the spectrum analyzer feature, you see two peaks at opposite ends of each other.

jopohl, 2018-07-29:
The mirroring is part of the nature of FFT, see e.g. [here](https://dsp.stackexchange.com/questions/4825/why-is-the-fft-mirrored) for a more detailed explanation. So what URH shows is the result of an FFT for spectrum analyzer and STFT for spectrogram, respectively.

So why does a SDR-Console captured signal does not show the negative side of the spectrum? I am not sure, maybe they perform a filter on the received data to eliminate negative sides of the spectrum.

vsboost, 2018-07-29:
All good then, i'm just not used to seeing it.
Ill close this one off.

closed 2018-07-29 by vsboost

transcats, 2019-06-01:
> The mirroring is part of the nature of FFT, see e.g. [here](https://dsp.stackexchange.com/questions/4825/why-is-the-fft-mirrored) for a more detailed explanation. So what URH shows is the result of [...]

You're thinking of the DFT (or FFT) of a real-only input. DFT of _n_ complex I/Q samples gives you an _n_ Hz spectrum with no mirroring (at least in theory; you can get some mirroring due to imperfections in the radio).

This is a bug which appears to have existed since native SDRPlay support was added and still exists in the current 2.6.0 release, and was inadvertently fixed by recent commits addressing #652, which seems to have corrected the interpretation of the I/Q samples for SDRPlay.

@vsboost, if you still care, try again either running from source (a bit annoying to set up) or wait for the next release.