jopohl/urh #521: Tests failing

FFY00, 2018-08-26:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
For urh to pass all the tests, even with `xvfb-run`.

## Actual Behavior
When running the tests in my local installation some test fail as you can see in the following log http://ix.io/1lj6.
The tests should also pass in a clean chroot, as we don't have a graphical interface there, we use `xvfb-run` to run the test command in a virtual X environment. The problem with this is that some tests fail with a core dump and the following message `DISPLAY=:$SERVERNUM XAUTHORITY=$AUTHFILE "$@" 2>&1`.

## Steps to Reproduce the Problem

###### Local machine
  1. `python setup.py pytest` (should work the same as calling pytest directly)

###### Clean chroot
  1. `xvfb-run python setup.py pytest`

## Platform Specifications

  - Python Version: 3.7
  - Operating System: Archlinux
  - Version of URH: 2.2.3
  - URH was installed: Github tarball

jopohl, 2018-08-27:
I could reproduce some failing tests on arch linux. Just merged a fix for that. Can you test latest master on your end? If everything works I can publish a new release.

FFY00, 2018-08-27:
I still get a segfault when running the tests I mentioned and the commit only fixed one file.

http://ix.io/1lml

jopohl, 2018-08-27:
You need to install pytest-forked plugin, otherwise you will experience multiprocessing errors which lead to the segfaults you mentioned:
```
pip3 install pytest-xdist
````

FFY00, 2018-08-27:
Didn't solve the issue.

```
plugins: xdist-1.23.0, forked-0.2
```

```
tests/test_continuous_modulator.py /usr/bin/xvfb-run: line 181:  1620 Segmentation fault      (core dumped) DISPLAY=:$SERVERNUM XAUTHORITY=$AUTHFILE "$@" 2>&1
```

jopohl, 2018-08-27:
If you have these two installed, please run pytest with the ```  --boxed ``` option for better debugging.

```
xvfb-run pytest --boxed tests/
````

FFY00, 2018-08-27:
Seems like a problem when compiling the cython modules http://ix.io/1lnb. Here's the tree http://ix.io/1lnd.

jopohl, 2018-08-27:
You could enforce building the extensions before running the tests:
```
python src/urh/cythonext/build.py
```

FFY00, 2018-08-29:
Works with https://github.com/jopohl/urh/commit/aac089f23824b31e62d355036b81ecb10ec1058d, can you create a release?

FFY00, 2018-08-30:
Released!

https://www.archlinux.org/packages/community/x86_64/urh/

closed 2018-08-30 by FFY00

jopohl, 2018-09-02:
Very nice!