jopohl/urh #529: RFE: Support multiple devices of the same type (Specifically rtl-sdr)

larsks, 2018-09-10:
I have multiple RTL-SDR dongles attached to my system. I would like to be able to specify which one urh should use.  At the moment, urh does not provide any mechanism for selecting a specific rtl-sdr device. If another program is using the "first" dongle, urh is unable to use a different device.

jopohl, 2018-09-10:
Are you using the latest version of URH? If so, you should be able to choose a RTL-SDR using a device identifier:

[image]

Once you have multiple RTL-SDRs attached, simply hit the green refresh button and you will be able to choose an identifier from the dropdown menu.

larsks, 2018-09-10:
My bad, I wasn't using the latest version. Sorry for the noise!

closed 2018-09-10 by larsks