jopohl/urh #533: Can't open libsoapySupport.so

FFY00, 2018-09-14:
## Expected Behavior
No errors.

## Actual Behavior
I get the following error.
```
Error: EnvironmentError: OSError: dlopen failed to load "/usr/lib/uhd/modules/libsoapySupport.so"
```
The file is there and has the permission table set `755` so this seems weird.

## Steps to Reproduce the Problem

  1. Install urh and soapyuhd in Arch Linux

## Platform Specifications

  - Python Version: 3.7.0
  - Operating System: Arch Linux
  - Version of URH: 2.2.4
  - URH was installed: repository

jopohl, 2018-09-14:
Interesting, I never saw this error before. Can't reproduce it either, even after installing soapyuhd. Can you be more specific when this error appears? During build or when using URH e.g. during recording a signal with USRP?

Nevertheless, the error should not affect URH in any way. I assume it comes from libuhd, complaining about the missing soapy support. Since URH does not rely on soapysdr in any way, everything should run as expected.

FFY00, 2018-09-14:
Hum. It just pops up when you start urh.

jopohl, 2018-09-14:
Can't reproduce this. Here is what I did on a fresh base/archlinux Docker Container:
```
pacman -S soapyuhd git cython python-setuptools python-numpy python-psutil python-pyzmq python-pyqt5 gcc
git clone https://github.com/jopohl/urh
cd urh/src/urh/
./main.py
... (Builds Extensions, including UHD support)
(Close an run again, just to be sure)
./main.py
```
The error you mentioned does not appear.

jopohl, 2018-09-14:
I'll close this one then.
1. The error cannot be reproduced on a fresh archlinux docker container.
2. The error is not significant anyway. URH does not depend on UHD's libsoapySupport at all.

closed 2018-09-14 by jopohl