jopohl/urh #536: LimeSDR in Docker

1am, 2018-09-17:
Hi,

I'm trying to use LimeSDR in URH from docker but it seems not to be working with spectrum analyser so far. It works ok on my host (Ubuntu 18) when I start the pip3 installed URH.

My Docker symptoms are the following:
* Device is detected and I can select it from the list
* After setting correct scanning parameters nothing happens, no error is thrown in terminal.
* Red LED on LimeSDR mini doesn't go off while on host it goes away and only green LED blinks.

Is it possible that it's a UI issue with X forwarding to Docker or maybe I should install all Lime tools inside of the docker container? Anybody experienced this issue before? 
If I remember correctly I had no issues with HackRF with Dockerized URH.

This is my URH SA on host
[image]

This is my URH SA in Docker
[image]

I'm not trying to run both at the same time :)

jopohl, 2018-09-17:
Strange, I cannot reproduce this behaviour. My LimeSDR runs fine in both, native and docker. I have, however, no LimeSDR mini for testing, just a normal one, so that might be the issue. Since the Docker Image is based on Ubuntu 18 I am very surprised, that the Lime does not work in your case. What version if LimeSuite you have installed on the host? In Docker Image it is 17.12.

1am, 2018-09-17:
My hos LimeUtil gives the following output:

```
LimeUtil --info   
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v18.04.1-myriadrf1~bionic
  Build timestamp:	2018-04-29
  Interface version:	v2018.3.0
  Binary interface:	18.04-1
...
Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
```

When I opened bash in urh image i couldn't find LimeUtil command so I've searched for it and have the following output:
```
# find / -name '*Lime*'
/usr/local/lib/python3.6/dist-packages/urh/dev/native/__pycache__/LimeSDR.cpython-36.pyc
/usr/local/lib/python3.6/dist-packages/urh/dev/native/LimeSDR.py
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.18.06-1
/usr/lib/x86_64-linux-gnu/libLimeSuite.so
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12.0
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.18.06.0
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12-1
```

Where on my host it's:

```
...
/usr/lib/x86_64-linux-gnu/cmake/LimeSuite
/usr/lib/x86_64-linux-gnu/cmake/LimeSuite/LimeSuiteConfig.cmake
/usr/lib/x86_64-linux-gnu/cmake/LimeSuite/LimeSuiteConfigVersion.cmake
/usr/lib/x86_64-linux-gnu/libLimeSuite.so
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.18.04.1
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.18.04-1
/usr/lib/x86_64-linux-gnu/pkgconfig/LimeSuite.pc
...
```

So maybe the coexistence of 18.06 and 17.12 could cause problems? 

Looked into the [Dockerfile](https://github.com/jopohl/urh/blob/master/data/Dockerfile) and can't find the LimeUtil installation part. I'm using the `FROM jopohl/urh:latest` in my own Dockerfile which doesn't have anything more yet so the source image is downloaded from DockerHub.

jopohl, 2018-09-17:
Thanks for your research. I just saw that limesuite17.12 is installed in the docker container. Does it help if you run apt-get remove limesuite17.12 and start URH after that?

jopohl, 2018-09-17:
Update: I just made some tweaks in the Dockerfile. Could you build it locally (available here: https://github.com/jopohl/urh/pull/537) and see if the Lime mini works? Normal LimeSDR works fine on my end.

1am, 2018-09-18:
I've tested the new dockerfile and there is a change, though it hasn't started correctly. Now I've got this error in the SA UI:

```
LimeSDR-OPEN (LimeSDR Mini, media=USB 2.0, module=uLimeSDR, addr=24607:1027, serial=1D424078480CB5): Success
LimeSDR-INIT: Error Code: -1 (-1)
LimeSDR: failed to start rx mode
```

In terminal I'm getting this message which I don't really know where to look at for solution:

```
[INFO::SettingsProxy.py::get_receive_buffer_size] Try to allocate receive buffer with size 262.144kB
[INFO::Device.py::start_rx_mode] LimeSDR: Starting RX Mode
Claimed Interface
Estimated reference clock 40.0018 MHz
Selected reference clock 40.000 MHz
[INFO::Device.py::log_retcode] LimeSDR-OPEN (LimeSDR Mini, media=USB 2.0, module=uLimeSDR, addr=24607:1027, serial=1D424078480CB5): Success
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=63, FRAC=0, DIV_OUTCH_CGEN=15
[ERROR::Device.py::log_retcode] LimeSDR-INIT: Error Code: -1 (-1)
libusb: warning [libusb_exit] application left some devices open
```

It doesn't seem that you changes even touched this part.

jopohl, 2018-09-18:
Alright, seems like we are getting somewhere. I just pushed another update to the Dockerfile. Now I use the liblimesuite version of the myriadrf repos, which should exactly match the version on your host.

Could you test it again?

1am, 2018-09-18:
Works like a charm!
Thank you

commit a08e9c0d (2018-09-18): update dockerfile; fix #536

* update dockerfile, possible fix for #536

* use myriadrf repos for installing limesuite
closed 2018-09-18 by 1am