jopohl/urh #545: Can't recognize RTL-SDR device

fenyvesi, 2018-10-03:
##### Expected Behavior
Wanted to capture in Spectrum A. 

##### Actual Behavior
1. Device Field: RTL-SDR
2. Device Identifier field is empty, can't select anything. (In Windows Device Manager there is an USB device named RTL2832U.)
3. If I push the green circle-shaped arrow next to the device id. selection, then the program doesn't react anymore, has to be aborted.
4. If I start capturing, the error message is: RTLSDR-OPEN (#0) Error Code: -1, RTLSDR failed to start RX mode

##### Steps To Reproduce
1. URH v2.3.0. Windows x64 installed
2. Zadig used for replace WIN driver, actual driver version: WINUSB (v6.1.7600.16385), USB Id: 0BDA 2832, device name: RTL2832U
3. SDR# works with the dongle and driver.

##### Screenshots

##### Platform Specifications
 - OS: WIN 10 x64
 - URH version: v2.3.0
 - Python version: 2.7
 - Installed via msi

jopohl, 2018-10-03:
Just checked RTL-SDR on my Windows VM and I could not reproduce this error. 

Could this issue #389 be related to your problem? There is also a solution given in the thread.

fenyvesi, 2018-10-03:
I saw #389 issue also, but I am afraid this is a different story.

1. I installed the driver for the composite device (RTL2832U) in Zadig.
2. My error code is not 12 but -1.
3. SDR# works. I suggest it uses this driver but I'm not sure.

I hesitate what should I do. Any advice?

fenyvesi, 2018-10-03:
I installed the URH 2.3.0 32 bit version  on another WIN 10 x86 (32 bit) machine, set USB driver with Zadik. The whole procedure was the same as in the previous x64 case.

URH is able to reach the dongle and works.

Somehow the difference seems to be in either WIN10 x64 or USB x64. I can't decide.

jopohl, 2018-10-03:
This is really mysterious. :thinking: There shouldn't be any difference in x32 and x64. 

Do you have access to another fresh WIN 10 (64 bit) machine? VM for example? I am starting to believe, your Win 10 x64 machine is affected by some Windows USB configuration voodoo. As said, I cannot reproduce the error in a x64 VM, although I use a Windows 7 x64 VM...

fenyvesi, 2018-10-04:
Yes, I will try it today.

fenyvesi, 2018-10-04:
Just a question if you know by chance:
Does SDR# use the same installed WInUSB driver  or its own solution? 
If they use the same driver, then the problem is in URH x64. If not, then there is no answer, yet.
I will try it also.

fenyvesi, 2018-10-04:
1. I attached the dongle to a new Win10 x64 machine
2. Replaced driver with Zadig
3. The USB device RTL2832U is in Device Manager
4. Neither SDR# nor URH is able to reach the device.
5. I tried to switch off the Fast Boot option of Windows (adivised to improve device management). This is a woodoo. Reboot. No result.
6. Conclusion: the reason of the problem seems to be the total confusion of device/driver management of Windows10.

closed 2018-10-04 by fenyvesi