jopohl/urh #548: New function in decoding (base) Time domain separator

archangelninja, 2018-10-09:
I'm trying to use the functions to decode a "simple" home automation protocol  (Nexa), the protocol used is a version of OOK,  where a "1" is pulse for T and no pulse for T, and a "0" is a pulse for T and no pulse for 5T
T= 250 us
5T = 1250 us 
More info here: http://tech.jolowe.se/home-automation-rf-protocols/

So Feature request: 
Add support for a decode function that uses the time domain as a separator between pulses.
So for example use the following:
- Edge trigger 0->1
- Then "sense" time X
- Trigger 1->0
- Then sense time until next edge trigger 0->1
if X= T, decode "1"
if X=5T, decode "0"

It would be great if it was possible to use the graphical interface to "draw" and select what should be "1" or "0". I.e. when in decoding window use the upper graph "window" to select the area that should be a "0" or "1" and then in the lower graph "window" define the time and the value to decode. But if you do this you need to "lock" the zoom/timescale between the upper and the lower graph "window".

Attached a zip with a sample file.
[HackRF-433_984MHz-2MSps-2MHz.complex.zip](https://github.com/jopohl/urh/files/2461944/HackRF-433_984MHz-2MSps-2MHz.complex.zip)

andynoack, 2018-10-10:
Hey @archangelninja,
actually the decoding can already be done with the current version of URH. We do not need a time dependent evaluation, as you can set the bitlength to 250µs (bitlength=500 @2MS/s) and work with the symbols (1/0) created by URH.

Let me sketch how to do it!
- Choose a bitlength of 500 for your signal
- Choose noise and center appropriately, I have tested with noise = 0.035 and center = 0.0398 (but I do not know, how the resulting values should look like).
- Then you need to craft a custom decoding 'Nexa'. However, there is need for finetuning, so this is just an idea:

1. Invert
// This signal is changed from ...0100001010000... to ...101111101011111... which is necessary because every symbol 0/1 begins with a high pulse (-> doesn't matter) and is followed by one or five zero pulses. Idea: The invert creates '1' -> 0 and '11111' -> 1.
2. Morse Code with
- Maximum length of 1-sequence for Low(0) = 2
- Minimum length of 1-sequence for High(1) = 4
- Numbers of 0s between 1 -sequences = 1
// '1' is converted to 0, '11111' is converted to 1
3. Invert
// As you like it the other way around, swap 0 and 1
4. Edge Trigger
// Finally you like to convert 01->0 and 10->1 which can be done by either Edge Trigger or a Substitution.

For me there are some decoding errors as the signal is not clean with current noise/center configuration. Maybe it helps to add some labels in NRZ decoding and configure them to not apply a decoding. After that switch the whole signal to 'Nexa' (in analysis).

Issue resolved?
Best Andreas

archangelninja, 2018-10-10:
Awesome! Thank you for this! Now I know how to calc it, I'll do some fiddling, will test this when the kids are sleeping! 
Issue Solved!
Best Richard

closed 2018-10-10 by archangelninja