jopohl/urh #554: Two issues on Linux

bluegizmo83, 2018-10-20:
Hi, I'm back again lol :)

I just installed URH on my Linux system (Parrot OS), and I have two issues with it so far. I installed it via sudo pip3 install urh, and then also tried updating via pip3 install --upgrade urh.

1. I'm on version 2.4.0 but I don't see the DC Correction option in the live recording options (it does show up as a filter option after recording though!)
2. I can't use the Spectrum Analyzer on the Linux install for some reason. As soon as I hit start, URH immediately closes completely. Record Signal still works fine though.

jopohl, 2018-10-20:
Welcome back :smile: 

I think both of your issues are caused by the same problem:
1. The DC Correction is only available for devices with __native backend__ enabled
2. Switching to native backend will most likely also fix the crash

So just go to Edit -> Options -> Device and switch to native backend for your SDR.

jopohl, 2018-10-21:
You are missing the airspy.h file. Assuming that you are using ubuntu, did you install the ``` libairspy-dev ``` package?

jopohl, 2018-10-24:
I am closing here due to missing response. Just drop a comment if you want this issue reopened.

closed 2018-10-24 by jopohl