jopohl/urh #577: USRP Segmentation Fault

ftwang, 2018-11-28:
Hi, I'm having a bit of trouble getting the USRP working with URH on my Mac.  Any help or guidance is appreciated!  After compiling the native extension, when using the URH with the USRP native device will result in a segmentation fault.  

##### Expected Behavior
Expect TestUSRP.py to pass unit test.

##### Actual Behavior
When running unit test for USRP, segmentation fault occurs when calling usrp.setup_stream()

##### Steps To Reproduce
1. Install UHD: sudo port install uhd
2. Clone Repository: git clone https://github.com/jopohl/urh/
3. Install with Native USRP support: python3 setup.py --with-usrp install
4. Run Unit test: python3 TestUSRP.py
5. Segmentation fault

##### Screenshots
> python3 TestUSRP.py                           1 ↵  11288  22:51:59
[INFO] [UHD] Mac OS; Clang version 8.1.0 (clang-802.0.38); Boost_106600; UHD_3.13.0.2-MacPorts-Release
1
['type=b200,name=,serial=xxxxxx,product=B200']
open 11
[1]    25205 segmentation fault  python3 TestUSRP.py

##### Platform Specifications
 - OS: Mac 10.12.6
 - Memory: 16 GB
 - URH version: 2.5.1
 - Python version: 3.7.1
 - Installed from source. 
 - SDR: Ettus B200 (over USB3.0)

jopohl, 2018-12-02:
I have neither a Mac nor a B200 to test this, but the unittest indicates that the open procedure failed. Does it help when you replace the line ``` return_code = usrp.open("addr=192.168.10.2") ``` in ``` TestUSRP.py ``` with  ``` return_code = usrp.open("") ``` or ``` return_code = usrp.open("type=b200") ```?

ftwang, 2018-12-02:
Thanks for responding.

If I replace with the following **return_code = usrp.open("")**

```
[INFO] [UHD] Mac OS; Clang version 8.1.0 (clang-802.0.38); Boost_106600; UHD_3.13.0.2-MacPorts-Release
[ERROR] [UHD] Device discovery error: send_to: No route to host
[ERROR] [USRP2] USRP2 Network discovery error send_to: Host is down
[ERROR] [E300] E300 Network discovery error send_to: Host is down
[ERROR] [E300] E300 Network discovery error send_to: Host is down
[ERROR] [X300] X300 Network discovery error send_to: Host is down
[ERROR] [UHD] Device discovery error: send_to: Host is down
1
['type=b200,name=,serial=F57256,product=B200']
[ERROR] [UHD] Device discovery error: send_to: Host is down
[ERROR] [USRP2] USRP2 Network discovery error send_to: Host is down
[ERROR] [E300] E300 Network discovery error send_to: Host is down
[ERROR] [E300] E300 Network discovery error send_to: Host is down
[ERROR] [X300] X300 Network discovery error send_to: Host is down
[ERROR] [UHD] Device discovery error: send_to: Host is down
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
open 0
[INFO] [B200] Asking for clock rate 32.000000 MHz...
[INFO] [B200] Actually got clock rate 32.000000 MHz.
Made rx_streame handler
Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2

Set sample rate 0
Set freq 0
Set bandwidth 0
Set gain 0
^CTraceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/connection.py", line 200, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/connection.py", line 398, in _send_bytes
    self._send(buf)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
KeyboardInterrupt
Exception ignored in: 'urh.dev.native.lib.usrp.recv_stream'

```
If I replace with **return_code = usrp.open("type=b200")**
```
[INFO] [UHD] Mac OS; Clang version 8.1.0 (clang-802.0.38); Boost_106600; UHD_3.13.0.2-MacPorts-Release
1
['type=b200,name=,serial=F57256,product=B200']
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
open 0
[INFO] [B200] Asking for clock rate 32.000000 MHz...
[INFO] [B200] Actually got clock rate 32.000000 MHz.
Made rx_streame handler
Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2

Set sample rate 0
Set freq 0
Set bandwidth 0
Set gain 0
[1]    75984 segmentation fault  python3 TestUSRP.py
```

jopohl, 2018-12-02:
Okay, the ``` return_code = usrp.open("") ``` seems to work and this should also be used in the Main GUI if you record/send with USRP. Does it work there?

ftwang, 2018-12-02:
Both with the change usrp.open("") and usrp.open("type=b200"), still results in SegFault in the unit test.  Now the SegFault occurs on this line:

```usrp.recv_stream(child_conn, num_samples)```

```
Python 3.7.1 (v3.7.1:260ec2c36a, Oct 20 2018, 03:13:28)
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from multiprocessing.connection import Pipe
>>>
>>> import sys
>>>
>>> from urh.util import util

util.set_shared_library_path()

from urh.dev.native.USRP import USRP
from urh.dev.native.lib import usrp
import unittest
>>>
>>> util.set_shared_library_path()
>>>
>>>
>>> from urh.dev.native.USRP import USRP
>>> from urh.dev.native.lib import usrp
>>> import unittest
>>> usrp.set_tx(False)
>>> return_code = usrp.open("")
[INFO] [UHD] Mac OS; Clang version 8.1.0 (clang-802.0.38); Boost_106600; UHD_3.13.0.2-MacPorts-Release
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
>>> usrp.setup_stream()
[INFO] [B200] Asking for clock rate 32.000000 MHz...
[INFO] [B200] Actually got clock rate 32.000000 MHz.
0
>>> usrp.set_sample_rate(2e6)
0
>>> usrp.set_center_freq(433.92e6)
0
>>> usrp.set_bandwidth(1e6)
0
>>> usrp.set_rf_gain(0.5)
0
>>> buffer = bytearray()
>>>
>>> num_samples = 32768 // 2
>>> usrp.start_stream(num_samples)
0
>>> parent_conn, child_conn = Pipe()
>>> usrp.recv_stream(child_conn, num_samples)
[1]    82617 segmentation fault  python3
```

jopohl, 2018-12-02:
Ah I see. Should not take about 5 minutes for sure. Can you test on another OS like Linux for example?

andynoack, 2018-12-13:
Closed due to inactivity, comment if you like the issue to be reopened.

closed 2018-12-13 by andynoack