jopohl/urh #578: Can't install on Ubuntu 16.04

nosejam, 2018-12-01:
##### Expected Behavior
Expected to install.

##### Actual Behavior
Fails to install.  Final error is: 
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

##### Steps To Reproduce
Install all prerequisites.
sudo pip3 install urh

##### Screenshots
`Collecting urh
  Using cached https://files.pythonhosted.org/packages/a7/ed/99d8c330c0e8b79fe6750df39708cc0a4a19b5f26fea9961c1c0e4a34501/urh-2.5.1.tar.gz
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from urh) (1.11.0)
Requirement already satisfied: psutil in /usr/lib/python3/dist-packages (from urh) (3.4.2)
Requirement already satisfied: pyzmq in /usr/lib/python3/dist-packages (from urh) (15.2.0)
Requirement already satisfied: cython in /usr/lib/python3/dist-packages (from urh) (0.23.4)
Installing collected packages: urh
  Running setup.py install for urh ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-g46qgaxf/urh/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-b0xitmqh/install-record.txt --single-version-externally-managed --compile:
    Found iio lib. Will compile with native plutosdr support
    Found rtlsdr lib. Will compile with native rtlsdr support
    Found airspy lib. Will compile with native airspy support
    Found uhd lib. Will compile with native usrp support
    Found LimeSuite lib. Will compile with native limesdr support
    Found mirsdrapi-rsp lib. Will compile with native sdrplay support
        Detected SDRPLAY_API_VERSION v2.1
    Found hackrf lib. Will compile with native hackrf support
    Found bladeRF lib. Will compile with native bladerf support
        Detected BLADERF_API_VERSION v1.9
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.5
    creating build/lib.linux-x86_64-3.5/urh
    copying src/urh/version.py -> build/lib.linux-x86_64-3.5/urh
    copying src/urh/main.py -> build/lib.linux-x86_64-3.5/urh
    copying src/urh/constants.py -> build/lib.linux-x86_64-3.5/urh
    copying src/urh/colormaps.py -> build/lib.linux-x86_64-3.5/urh
    copying src/urh/__init__.py -> build/lib.linux-x86_64-3.5/urh
    package init file 'src/urh/__pycache__/__init__.py' not found (or not a regular file)
    creating build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/util.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/__init__.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/WSPChecksum.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/SettingsProxy.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/RingBuffer.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/ProjectManager.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/Logger.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/HTMLFormatter.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/GenericCRC.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/Formatter.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/FileOperator.py -> build/lib.linux-x86_64-3.5/urh/util
    copying src/urh/util/Errors.py -> build/lib.linux-x86_64-3.5/urh/util
    creating build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/xtra_icons_rc.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/urh_rc.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_tab_interpretation.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_simulator_dialog.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_simulator.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_signal_frame.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_signal_details.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_send_recv_sniff_settings.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_send_recv_device_settings.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_send_recv.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_properties_dialog.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_project.py -> build/lib.linux-x86_64-3.5/urh/ui
    copying src/urh/ui/ui_plugins.py -> build/lib.linux-x86_64-3.5/urh/ui
[... 508 lines cut ...]
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.5/src/urh/dev/native/lib/limesdr.o -lLimeSuite -o build/lib.linux-x86_64-3.5/urh/dev/native/lib/limesdr.cpython-35m-x86_64-linux-gnu.so
    building 'urh.dev.native.lib.sdrplay' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/usr/lib/python3/dist-packages/numpy/core/include -c src/urh/dev/native/lib/sdrplay.cpp -o build/temp.linux-x86_64-3.5/src/urh/dev/native/lib/sdrplay.o -Wno-cpp
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    src/urh/dev/native/lib/sdrplay.cpp: In function ‘PyObject* __pyx_f_3urh_3dev_6native_3lib_7sdrplay_init_stream(int, double, double, double, double, PyObject*, int)’:
    src/urh/dev/native/lib/sdrplay.cpp:3941:455: error: invalid conversion from ‘void (*)(short int*, short int*, unsigned int, int, int, int, unsigned int, unsigned int, unsigned int, void*)’ to ‘mir_sdr_StreamCallback_t {aka void (*)(short int*, short int*, unsigned int, int, int, int, unsigned int, unsigned int, void*)}’ [-fpermissive]
       __pyx_t_1 = __Pyx_PyInt_From_mir_sdr_ErrT(mir_sdr_StreamInit((&__pyx_v_gain_reduction), (__pyx_v_sample_rate / 1e6), (__pyx_v_center_freq / 1e6), __pyx_v_bw_type, __pyx_v_if_type, __pyx_v_lna_state, (&__pyx_v_gRdBsystem), __pyx_v_3urh_3dev_6native_3lib_7sdrplay_set_gr_mode, (&__pyx_v_samples_per_packet), __pyx_f_3urh_3dev_6native_3lib_7sdrplay__rx_stream_callback, __pyx_f_3urh_3dev_6native_3lib_7sdrplay__gain_change_callback, ((void *)__pyx_v_func))); if (unlikely(!__pyx_t_1)) __PYX_ERR(0, 154, __pyx_L1_error)
                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ^
    In file included from src/urh/dev/native/lib/sdrplay.cpp:642:0:
    /usr/local/include/mirsdrapi-rsp.h:288:36: note:   initializing argument 10 of ‘mir_sdr_ErrT mir_sdr_StreamInit(int*, double, double, mir_sdr_Bw_MHzT, mir_sdr_If_kHzT, int, int*, mir_sdr_SetGrModeT, int*, mir_sdr_StreamCallback_t, mir_sdr_GainChangeCallback_t, void*)’
        _MIR_SDR_QUALIFIER mir_sdr_ErrT mir_sdr_StreamInit(int *gRdB, double fsMHz, double rfMHz, mir_sdr_Bw_MHzT bwType, mir_sdr_If_kHzT ifType, int LNAstate, int *gRdBsystem, mir_sdr_SetGrModeT setGrMode, int *samplesPerPacket, mir_sdr_StreamCallback_t StreamCbFn, mir_sdr_GainChangeCallback_t GainChangeCbFn, void *cbContext);
                                        ^
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-g46qgaxf/urh/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-b0xitmqh/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-g46qgaxf/urh/
`

##### Platform Specifications

 - OS: Ubuntu 16.04.5
 - URH version: 2.5.1
 - Python version: 3.5.2
 - Installed via pip

jopohl, 2018-12-01:
Does it work when you update you SDRPlay driver to the [most recent version](https://www.sdrplay.com/downloads/)?

nosejam, 2018-12-02:
Finally installed.  Not sure what is different.  Multiple reboots and reinstallation of dependencies.  Thanks for your help.

closed 2018-12-02 by nosejam