jopohl/urh #580: MacOS install fail

jeffg2k, 2018-12-06:
Install fails

> In file included from src/urh/cythonext/signal_functions.cpp:25:
> /usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:14:2: error: 
>       "Something's broken. UCHAR_MAX should be defined in limits.h."
> #error "Something's broken.  UCHAR_MAX should be defined in limits.h."
>  ^
> /usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:18:2: error: 
>       "Python's source code assumes C's unsigned char is an 8-bit type."
> #error "Python's source code assumes C's unsigned char is an 8-bit type."
>  ^
> In file included from src/urh/cythonext/signal_functions.cpp:25:
> In file included from /usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:34:
> In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdlib.h:94:
> In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdlib.h:66:
> In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/wait.h:110:
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:196:2: error: 
>       unknown type name 'uint8_t'
>         uint8_t  ri_uuid[16];
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:197:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_user_time;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:198:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_system_time;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:199:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_pkg_idle_wkups;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:200:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_interrupt_wkups;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:201:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_pageins;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:202:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_wired_size;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:203:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_resident_size;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:204:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_phys_footprint;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:205:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_proc_start_abstime;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:206:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_proc_exit_abstime;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:210:2: error: 
>       unknown type name 'uint8_t'
>         uint8_t  ri_uuid[16];
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:211:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_user_time;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:212:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_system_time;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:213:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_pkg_idle_wkups;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:214:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_interrupt_wkups;
>         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/resource.h:215:2: error: 
>       unknown type name 'uint64_t'
>         uint64_t ri_pageins;
>

jopohl, 2018-12-06:
Is this workaround from @peterg10 helping you?

```
MACOSX_DEVELOPMENT_TARGET=10.14 pip3 install --upgrade urh
```

jopohl, 2018-12-07:
Furthermore, the above errors
```
/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:14:2: error:
"Something's broken. UCHAR_MAX should be defined in limits.h."
#error "Something's broken. UCHAR_MAX should be defined in limits.h."
^
/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m/Python.h:18:2: error:
"Python's source code assumes C's unsigned char is an 8-bit type."
#error "Python's source code assumes C's unsigned char is an 8-bit type."
```
indicate, there is something wrong with your Python installation on your system.

jeffg2k, 2018-12-07:
That's really weird. I reinstalled Python and it worked.  Thanks for the assist.

closed 2018-12-07 by jeffg2k