jopohl/urh #581: CLI decoding

nosejam, 2018-12-08:
##### Expected Behavior
Using the command line interface continuously demodulate and decode a signal.  Expected the decoded data to differ from the demodulated data.

##### Actual Behavior
Decoded result returns the same bits as demodulated bits without decoding.

##### Steps To Reproduce
This command:
urh_cli.py -d RTL-SDR -di "Realtek RTL2838UHIDIR (SN: 00000001)" -db native -f 433.920e6 -s 1e6 -b 1e6 -g 199 -mo ASK -p0 0 -p1 1 -bl 500 -n 0.03 -c 0.16 -t 0 -e -rx -rt -1 -v

Returns data beginning with the expected preamble:
0011001100110011001100110011001100110011...

Adding my custom pulse interval width modulation (PIWM) decoding to the command returns the same thing.
Command:
urh/cli/urh_cli.py -d RTL-SDR -di "Realtek RTL2838UHIDIR (SN: 00000001)" -db native -f 433.920e6 -s 1e6 -b 1e6 -g 199 -mo ASK -p0 0 -p1 1 -bl 500 -n 0.03 -c 0.16 -t 0 -e PIWM -rx -rt -1 -v

Returns:
0011001100110011001100110011001100110011...

Using the same external decoding in the GUI returns the following (0011 => 11 in this decoding):
11111111111111111111...

Also attempting a non-custom decoding such as Manchester I returns the same result as no decoding.
Command:
urh_cli.py -d RTL-SDR -di "Realtek RTL2838UHIDIR (SN: 00000001)" -db native -f 433.920e6 -s 1e6 -b 1e6 -g 199 -mo ASK -p0 0 -p1 1 -bl 500 -n 0.03 -c 0.16 -t 0 -e "Manchester I" -rx -rt -1 -v

Returns:
0011001100110011001100110011001100110011...

Although, in the GUI, Manchester I returns:
101010101010101010...

##### Screenshots
N/A (see above)

##### Platform Specifications

    OS: Ubuntu 16.04.5
    URH version: 2.5.1
    Python version: 3.5.2
    Installed via pip

jopohl, 2018-12-08:
As documented [here](https://github.com/jopohl/urh/wiki/Command-Line-Interface#example) you need to configure the full decoding and not only give the name. You can find that out using e.g. 
```
$ cat ~/.config/urh/decodings.txt
'Non Return To Zero (NRZ)', 
'Non Return To Zero Inverted (NRZ-I)', 'Invert', 
'Manchester I', 'Edge Trigger', 
'Manchester II', 'Edge Trigger', 'Invert', 
'Differential Manchester', 'Edge Trigger', 'Differential Encoding',
```
so for Manchester I you would need to do ``` -e "'Arbitrary Name', 'Edge Trigger'" ```. In case you have a project file you can find your configured encoding in the ``` URHProject.xml ```.

closed 2018-12-08 by jopohl

nosejam, 2018-12-08:
Thanks!  This now works for the built in decodings (such as manchester), but I still can't get my external ones to work even though they work in the GUI.  
When I enter the external decoding exactly as it is listed in the ~/.config/urh/decodings.txt file, it returns an error: "[ERROR::util.py::run_command] Could not find"
I am entering  -e "'PIWM', 'External Program', '/home/nosejam/PIWM.py;'" which is exactly how it is listed in the decodings.txt file.  What am I doing wrong?

jopohl, 2018-12-08:
It works when you remove the single quotes (sorry for my misleading first answer here), so the correct argument in your case would be:

```
 -e "PIWM, External Program, /home/nosejam/PIWM.py;"
```

nosejam, 2018-12-08:
That works!  Thanks!
By the way, this software is the best I've ever used.  The user interface is perfect to quickly understand a signal and decode it with very little effort.  I'm amazed by how many signals I've been able to analyze and decode in just a few hours of playing with it.