jopohl/urh #582: Lora

zwlin, 2018-12-10:
Is it possible to add Lora protocol decoding

andynoack, 2018-12-10:
In general, LoRa is interesting for us! Unfortunately we do not have LoRa devices in our lab (for testing) and, to be honest, there is a large amount of work to do. Don't expect LoRa to be integrated too soon, unless someone helps us out :)
We are happy to accept external decodings or other helpful stuff!

ikarus23, 2018-12-10:
Maybe decoding LoRa is a bit overkill for URH. At least for now. Otherwise I would ask for 6LoWPAN, ZigBee, Z-Wave, Sigfox, HomeMatic, EnOcean, etc. as well ;)

However, if you want to decode LoRa with your SDR just have a look at https://github.com/rpp0/gr-lora
or https://github.com/BastilleResearch/gr-lora. Both projects aim at decoding LoRa with a SDR using GnuRadio. For the second one there also is a talk from Matt Knight at the Chaos Communication Congress: https://media.ccc.de/v/33c3-7945-decoding_the_lora_phy However, if I'm not mistaken, the first project is more active and complete. Not sure though.

Zarados, 2019-05-30:
Hi all, I have a little question with the decode of the signal.

I notice @ikarus23 suggested increase more decode way of the IOT.(6LoWPAN, ZigBee, Z-Wave, Sigfox, HomeMatic, EnOcean, etc.)

The encode way of z-wave is Manchester coding (Maybe I'm wrong.) and urh is included it.

I used urh to analysis the z-wave signal.

The result of the decode doesn't like I used Waving-Z with rtl-sdr. (I compare two result by hex.)

In short, Could urh decode the signal of z-wave? 

If not, tell me why please because i think urh include the decode way of the  Manchester.

URH is awesome project!!

Respectfully

andynoack, 2019-06-03:
As long as there is only one channel (frequency) and BFSK (2FSK) modulation, it should be possible to decode the signal properly. Maybe you have to try around with the decoding, i.e. invert or differential or ... primitives in the decoding chain.

gitdode, 2025-04-21:
Could I ask for the status of this (decoding LoRa), is there still an interest in that?
I have some devices (RFM95) and a simple [driver](https://github.com/gitdode/librfm95) for them, and recording a signal works just fine:

[image]

So, is there any way I could contribute?

andynoack, 2025-04-21:
Until now there have not been major efforts to integrate LoRa, though this is still a topic for us from a research perspective. 

So, what can you so?
- We would be happy about test data, i.e. raw signals along with their contents as bit/hex stream and their precise encoding.
- Even better would be, If you try to decode these raw signals and point us to problems in URHs Decoding/Encoding process if there are any.

gitdode, 2025-04-21:
What I can do for now is to set (besides carrier frequency, bandwidth, bit rate etc.):

- Spreading factor (6)7 - 12
- Forward error correction code rate 4/5 - 4/8
- Preamble length, i.e. 12 symbols
- Payload CRC on/off
- Add header yes/no:
  - Payload length
  - Forward error correction code rate
  - Payload CRC presence
- Payload

The simplest would probably be to send a packet with no header and no CRC, so just the preamble + payload.
Unfortunately, unlike in FSK mode, the preamble polarity is not known and cannot be set.

Would such signals be useful?

For the moment, I have no idea how to try to decode a raw LoRa signal.

andynoack, 2025-04-21:
Yes, I think so! Actually it would not be a bad idea to keep data and a CRC in, since then it might be easier to see whether URH is able to decode correctly.

gitdode, 2025-04-28:
Sorry for the delay, I was away - here come two initial signals.

[RTL-SDR-20250428_202.tar.gz](https://github.com/user-attachments/files/19947075/RTL-SDR-20250428_202.tar.gz)

The first (RTL-SDR-20250428_202200-868_6MHz-2MSps-62_5KHz) is with "Low Data Rate Optimization":

> Given the potentially long duration of the packet at high spreading factors the option is given to improve the robustness of the transmission to variations in frequency over the duration of the packet transmission and reception. The bit
LowDataRateOptimize increases the robustness of the LoRa link at these low effective data rates. Its use is mandated
when the symbol duration exceeds 16ms. Note that both the transmitter and the receiver must have the same setting for LowDataRateOptimize.

The other signal is without LDRO. I thought both variants might be interesting.

Packet structure:

[image]

- Preamble: 12 symbols
- Header (explicit header mode), transmitted with error correction code rate 4/8:
  - Payload length in bytes: 4 bytes
  - Payload forward error correction code rate: 4/5
  - Presence of an optional 16-bit CRC for the payload: yes
  - Header's own CRC (also 16-bit I suppose)
- Payload: `{0xde, 0xad, 0xbe, 0xef}`
- 16-bit payload CRC

Transmitter configuration:
- Signal bandwidth: 62.5 kHz
- Error correction code rate: 4/5
- Spreading factor: 9

I hope I did not forget any required info about the signals, and that they are useful at all.
Just let me know if I can send you more signals with other settings, payload, ... it is easy to record signals in URH ;-)

andynoack, 2025-04-30:
I had a deeper look into it and it is complicated! Chirp Code Demodulation (a special FSK variant) is not supported by URH and I am not sure whether it is possible to implement it without breaking out of the current method, how demodulation is implemented at the moment.
In theory, I think, that it should be possible to craft an encoding with the help of the substitution table function of URH. The provided signals are using a spreading factor of 9 which means that there must be 512 entries in the table...that is pretty much and I guess that the precision may not suffice to create a fault tolerant/stable decoding.

Would you create test signals with a spreading factor of 2 and 3? Maybe this helps to find a decoding algorithm that can be scaled to 9 (or whatever) later.

gitdode, 2025-04-30:
Oh I can sure imagine it is not trivial!

Unfortunately the RFM95 radios I have can only do SF 6-12, and 6 only in "Implicit Header Mode", so the packet is just the preamble, payload and CRC - but the simpler the better, or?

> Implicit Header Mode
> In certain scenarios, where the payload, coding rate and CRC presence are fixed or known in advance, it may be advantageous to reduce transmission time by invoking implicit header mode. In this mode the header is removed from the packet. In this case the payload length, error coding rate and presence of the payload CRC must be manually configured on both sides of the radio link.
> Note
> With SF = 6 selected, implicit header mode is the only mode of operation possible.
> 

So this packet is:

- Preamble: 12 symbols
- Payload: {0xde, 0xad, 0xbe, 0xef}
- 16-bit payload CRC

Transmitter configuration:

- Signal bandwidth: 62.5 kHz
- Error correction code rate: 4/5
- Spreading factor: 6

[RTL-SDR-20250430_203047-868_6MHz-2MSps-62_5KHz.complex16s.tar.gz](https://github.com/user-attachments/files/19982812/RTL-SDR-20250430_203047-868_6MHz-2MSps-62_5KHz.complex16s.tar.gz)

It looks simpler and more "readable" to me. Hopefully it is more useful.

andynoack, 2025-05-01:
I tried this one, but could not recognize something. Would you create some more signals with an easier payload?

0xde, 0xad, 0xbe, 0xef =>
d       e       a       d       b       e       e       f
1101    1110    1010    1101    1011    1110    1110    1111

Spreading factor 6 means 6 bit per symbol:

110111  101010  110110  111110  111011  11
55      42      54      62      59      3 (decimal)

I suggest a binary payload like: 111111 000000 111000 000111

gitdode, 2025-05-01:
Sure, here comes a signal identical to the previous one but with the payload you suggested:

`{0b11111100, 0b00001110, 0b00000111}`

I am sending you the signal with all 4 forward error correction coding rates 4/5, 4/6, 4/7 and 4/8, maybe that is helpful?

[RTL-SDR-20250501_23.tar.gz](https://github.com/user-attachments/files/20005763/RTL-SDR-20250501_23.tar.gz)

You probably know this page: https://revspace.nl/DecodingLora?
It is interesting because it relates to the RFM95 (SX1272) radio module that I am using.

gitdode, 2025-05-02:
What do you think about these ideas from https://revspace.nl/DecodingLora?

- For simplicity, the output length of the forward error coding parameter CR can be chosen to be equal to the symbol size SF. This way each 4-bit input nibble should result in exactly one output symbol. For example use CR=4/6 with SF=6, CR=4/7 with SF=7, CR=4/8 with SF=8,
- make the payload equal to the PRNG sequence, such that the whitening effect of the PRNG is cancelled out, possibly making analysis easier.
- vary the payload with a walking-bit sequence, so the shuffling order of the interleaver can be analysed.

andynoack, 2025-05-02:
I was not aware that there is a whitening. This, you are absolutely right, whitening has to be elimated first to be able to see anything.
If you know the whitening sequence for the payload, payload should be something like whitening XOR an easy sequence, whereby an easy sequence should be something that makes it easy to differentiate the symbols from each other, e.g. 000000, 111111 or 000000, 000001, 000010, 000100, 001000, 010000, 100000, 111111.

gitdode, 2025-05-02:
Sorry, probably I should have mentioned that earlier.

I'm trying to find out the sequence used for whitening; maybe it is the "9-bit LFSR used to generate a random sequence" used for whitening in FSK mode?

Once I know I will create & send you new signals.

gitdode, 2025-05-03:
Just a thought - the 3-byte payload plus the two CRC bytes whitened in FSK mode are:

`00000011 10001001 10111101 11001010 01111110`
`11111100 00001110 00000111 00101011 00101000` (no whitening)

If we can assume that the SX1276 does the same whitening in LoRa mode (same LFSR with same seed) as in FSK mode, this should be the actual payload + CRC in the LoRa signal.