jopohl/urh #584: Add device support for XTRX

zhovner, 2018-12-14:
XTRX is a versatile and powerfull SDR https://www.crowdsupply.com/fairwaves/xtrx
It would be nice to add it to UHR.

They have two modes: 
* kernel module for mPCIe connection
* userspace driver for user USB3-mPCIe adapter 

Here it the sources: https://github.com/xtrx-sdr
And first start howto https://github.com/xtrx-sdr/images

zhovner, 2019-01-02:
I've already started to porting  [libxtrx](https://github.com/xtrx-sdr/libxtrx/) into URH here https://github.com/zhovner/urh/ 
But I have no idea what i'm actually doing with all of this Cython stuff and try to figure out by looking at another `*.pyx` and `*.pxd` lib's and copy&pasting in same manner. 

This example application that using `libxtrx` https://github.com/xtrx-sdr/libxtrx/tree/master/examples/xtrx_fft can be used as reference.

I would appreciate any help.

zhovner, 2019-01-05:
Little quick start guide for those who maybe want to join me. @JessieAMorris @ducnguyenAstro @ccsh23 @ynour @mpbraendli 

1. Install XTRX libraries and header. You should have them at following path: 
`/usr/local/lib/libxtrx.so`  
sources at  `/usr/include/libxtrx`  

2. Install URH with all dependencies 

3. Clone my fork in separate directory https://github.com/zhovner/urh/

Hardware support in URH made by Cython wrappers of original C libraries. 
Here is the manual of this technique https://cython.readthedocs.io/en/latest/src/tutorial/clibraries.html

Here is an example for HackRF (libhackrf): 

* libhackrf header redefined (.pxd file) [src/urh/dev/native/lib/chackrf.pxd](https://github.com/jopohl/urh/blob/master/src/urh/dev/native/lib/chackrf.pxd)
* wrapper [src/urh/dev/native/lib/hackrf.pyx](https://github.com/jopohl/urh/blob/master/src/urh/dev/native/lib/hackrf.pyx)
* and final URH hackrf library [src/urh/dev/native/HackRF.py](https://github.com/jopohl/urh/blob/master/src/urh/dev/native/HackRF.py)

I write some XTRX wrappers: 

* xtrx_api.h redefined https://github.com/zhovner/urh/blob/master/src/urh/dev/native/lib/cxtrx.pxd
* wrapper with only one function https://github.com/zhovner/urh/blob/master/src/urh/dev/native/lib/xtrx.pyx

To build this Cython wrapper into binary use this script: `urh/src/urh/cythonext/build.py`  
In my fork it builds only XTRX lib ignoring others. 

For the current moment I can build it successfully but not sure if this one function even works :)

JessieAMorris, 2019-01-05:
Nice. Have you committed your changes on your branch? I've been adding some of the auxiliary stuff that was missing, i.e. device config, etc. Mostly still working on getting familiar with the urh codebase still.

zhovner, 2019-01-06:
>Have you committed your changes on your branch?

Yes. At this moment we need to work at [xtrx.pyx](https://github.com/zhovner/urh/blob/master/src/urh/dev/native/lib/xtrx.pyx) to add all necessary functions. I think the good first step is to implement receive only functionality.

JessieAMorris, 2019-01-06:
I've made a good start on that. I have device listing and device opening working at this point. I'm working on frequency setting, gain, etc. as well.

JessieAMorris, 2019-01-06:
Not sure if you want to give me permissions to push to your fork or if I should just fork yours.

JessieAMorris, 2019-01-06:
For now, I've just created a fork to be able to push my code and so you can see what I've done so far: https://github.com/JessieAMorris/urh

I did have to go into the options menu in urh and enable the XTRX portion to get it to show up in the devices.

zhovner, 2019-01-06:
>Not sure if you want to give me permissions to push to your fork or if I should just fork yours.

I've grant you access to my repo. Fell free to merge.

JessieAMorris, 2019-01-06:
@zhovner I just pushed up my changes. I have a working (I think) spectrum analyzer now. I'm not sure if the data is accurate yet, but it is showing something that seems reasonable ish. I think I need to figure out gain options next.

Error handling still seems to be pretty poor. It seems like the XTRX can get into a funky state pretty easily. I've found that setting the sample rate is needed before tuning the radio and subsequent sample rate changes are iffy at best.

I've found that the `test_xtrx` command in the `libxtrx/build/` director (in the actual `libxtrx` project) seems to reset it and get into a healthy state.

zhovner, 2019-01-07:
Great, will test in few days.

zhovner, 2019-01-07:
I've record a screencast of testing spectrum analyzer with RTL-SDR and XTRX for comparison.

RTL-SDR: https://files.catbox.moe/zwsob8.mp4
XTRX: https://files.catbox.moe/2sysxl.mp4

Looks like XTRX does not receive any data. Sometimes it shows some static signal when i press Start button, but no new signal after. But at least it found a correct device identifier :)

JessieAMorris, 2019-01-07:
Yeah, I can believe that.  I need to take a closer look at the data that the xtrx is outputting and make sure that I'm formatting it correctly for urh. I probably won't have any time until tomorrow to look at that.

m4ver1cks, 2019-02-22:
I have been able to use XTRX in URH via gnuradio. 
[image]

I'm using Manjaro so some steps could be different. 
0º You need to install gnuradio-osmosdr from xtrx branch and ensure that you are using the correct drivers. Yo can ensure launching the osmocom_fft and if in the line that shows all drivers loaded (in my computer "built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya xtrx") have the string xtrx you are lucky. 
1º I download the code from GIT. 
2º I copy the files hackrf_recv.py and hack_send.py to xtrx_recv.py and xtrx_send.py
3º Modify some lines in each new files. Replace "hackrf" to "xtrx", Replace 433000 to 433000000.
4º Modify the file dev/BackendHandler.py and add xtrx in the list of the hardware (    DEVICE_NAMES = ("AirSpy R2", "AirSpy Mini", "BladeRF", "FUNcube", "HackRF",
                    "LimeSDR", "PlutoSDR", "RTL-SDR", "RTL-TCP", "SDRPlay", "SoundCard", "USRP", "xtrx"))
5º Install the package python2-pyzmq

JessieAMorris, 2019-02-25:
I just pushed my fixes to @zhovner's master branch.