jopohl/urh #585: [Bug] No text on white background in macOS dark theme

zhovner, 2018-12-15:
I think it's a Qt issue, but don't know how to fix this.

Here how it's look like. When text is selected it can be seen.

[image]

jopohl, 2018-12-15:
Can't reproduce this. This is how it looks on a OSX 10.13 VM with Qt 5.10.1:
[image]

zhovner, 2018-12-15:
>This is how it looks on a OSX 10.13
I mean the system wide dark mode that was added in macOS 10.14 Mojave. I don't change any view settings in URH. In my example you can see that window borders are dark too.
But if I set dark theme directly in URH, the problem is gone.