jopohl/urh #587: fix python3 only support

ZeroChaos-, 2018-12-17:
##### Expected Behavior
I expect this to work

##### Actual Behavior
wild errors when urh does silly things.  Ideally, on a python3 installation I shouldn't see things like /usr/bin/env python2

##### Steps To Reproduce
zero@theprophet urh % grep python2 $(qlist urh)                                                                                                                               (git)-[master] 
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        if self.backend_handler.python2_exe:
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:            self.ui.lineEditPython2Interpreter.setText(self.backend_handler.python2_exe)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        self.ui.lineEditPython2Interpreter.editingFinished.connect(self.on_python2_exe_path_edited)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        self.ui.btnChoosePython2Interpreter.clicked.connect(self.on_btn_choose_python2_interpreter_clicked)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        self.ui.radioButtonPython2Interpreter.clicked.connect(self.on_radio_button_python2_interpreter_clicked)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        self.ui.lineEditPython2Interpreter.setText(self.backend_handler.python2_exe)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:                self.tr("Gnuradio is not installed or incompatible with python2 interpreter."))
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        self.backend_handler.python2_exe = self.ui.lineEditPython2Interpreter.text()
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:    def on_python2_exe_path_edited(self):
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:    def on_btn_choose_python2_interpreter_clicked(self):
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:        filename, _ = QFileDialog.getOpenFileName(self, self.tr("Choose python2 interpreter"), filter=dialog_filter)
/usr/lib64/python3.6/site-packages/urh/controller/dialogs/OptionsDialog.py:    def on_radio_button_python2_interpreter_clicked(self, checked: bool):
Binary file /usr/lib64/python3.6/site-packages/urh/controller/dialogs/__pycache__/OptionsDialog.cpython-36.opt-2.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/controller/dialogs/__pycache__/OptionsDialog.cpython-36.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/controller/dialogs/__pycache__/OptionsDialog.cpython-36.opt-1.pyc matches
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/airspy_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/bladerf_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/bladerf_send.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/funcube_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/hackrf_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/hackrf_send.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/rtl-sdr_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/sdrplay_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/usrp_recv.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/usrp_send.py:#!/usr/bin/env python2
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:            if os.path.isfile(os.path.join(gnuradio_dir, "gr-python27", "pythonw.exe")):
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:                self.python2_interpreter = os.path.join(gnuradio_dir, "gr-python27", "pythonw.exe")
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:                self.python2_interpreter = os.path.join(gnuradio_dir, "gr-python27", "python.exe")
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:            self.python2_interpreter = constants.SETTINGS.value("python2_exe", "")
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:        if not self.python2_interpreter:
/usr/lib64/python3.6/site-packages/urh/dev/gr/AbstractBaseThread.py:        options = [self.python2_interpreter, os.path.join(rp, filename),
Binary file /usr/lib64/python3.6/site-packages/urh/dev/gr/__pycache__/AbstractBaseThread.cpython-36.opt-2.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/dev/gr/__pycache__/AbstractBaseThread.cpython-36.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/dev/gr/__pycache__/AbstractBaseThread.cpython-36.opt-1.pyc matches
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:        python2_exe = constants.SETTINGS.value('python2_exe', '')
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:        if not python2_exe:
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            self.__python2_exe = self.__get_python2_interpreter()
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            constants.SETTINGS.setValue("python2_exe", self.__python2_exe)
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            self.__python2_exe = python2_exe
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:        constants.SETTINGS.setValue("python2_exe", self.__python2_exe)
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:    def python2_exe(self):
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:        return self.__python2_exe
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:    @python2_exe.setter
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:    def python2_exe(self, value):
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:        if value != self.__python2_exe:
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            self.__python2_exe = value
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            constants.SETTINGS.setValue("python2_exe", value)
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            if os.path.isfile(self.python2_exe) and os.access(self.python2_exe, os.X_OK):
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:                    self.gnuradio_is_installed = call('"{0}" -c "import gnuradio"'.format(self.python2_exe),
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:    def __get_python2_interpreter(self):
/usr/lib64/python3.6/site-packages/urh/dev/BackendHandler.py:            for prog in ["python2", "python2.exe"]:
Binary file /usr/lib64/python3.6/site-packages/urh/dev/__pycache__/BackendHandler.cpython-36.opt-2.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/dev/__pycache__/BackendHandler.cpython-36.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/dev/__pycache__/BackendHandler.cpython-36.opt-1.pyc matches
/usr/lib64/python3.6/site-packages/urh/ui/ui_options.py:        self.lineEditPython2Interpreter.setPlaceholderText(_translate("DialogOptions", "/usr/bin/python2"))
Binary file /usr/lib64/python3.6/site-packages/urh/ui/__pycache__/ui_options.cpython-36.opt-2.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/ui/__pycache__/ui_options.cpython-36.pyc matches
Binary file /usr/lib64/python3.6/site-packages/urh/ui/__pycache__/ui_options.cpython-36.opt-1.pyc matches

##### Platform Specifications
 - OS: Gentoo Linux
 - URH version: 2.5.2
 - Python version: 3.6
 - Installed via emerge (I am the maintainer of the gentoo package)

jopohl, 2018-12-17:
The python2 stuff is required for the GNU radio backend, which is __optional__. So what is the exact problem? Especially, please specify this line:
> wild errors when urh does silly things

blshkv, 2018-12-17:
Our user reported the following:
```
pentoo-usb ~ # urh
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Skipping native support for airspy
Found bladeRF lib. Will compile with native bladerf support
    Detected BLADERF_API_VERSION v2.02
Found hackrf lib. Will compile with native hackrf support
Skipping native support for limesdr
Found iio lib. Will compile with native plutosdr support
Found rtlsdr lib. Will compile with native rtlsdr support
Found uhd lib. Will compile with native usrp support
Skipping native support for sdrplay
Skipping native support for airspy
Found bladeRF lib. Will compile with native bladerf support
    Detected BLADERF_API_VERSION v2.02
Found hackrf lib. Will compile with native hackrf support
Skipping native support for limesdr
Found iio lib. Will compile with native plutosdr support
Found rtlsdr lib. Will compile with native rtlsdr support
Found uhd lib. Will compile with native usrp support
Skipping native support for sdrplay
Compiling /usr/lib64/python3.6/site-packages/urh/dev/native/lib/bladerf.pyx because it changed.
Compiling /usr/lib64/python3.6/site-packages/urh/dev/native/lib/hackrf.pyx because it changed.
Compiling /usr/lib64/python3.6/site-packages/urh/dev/native/lib/plutosdr.pyx because it changed.
Compiling /usr/lib64/python3.6/site-packages/urh/dev/native/lib/rtlsdr.pyx because it changed.
Compiling /usr/lib64/python3.6/site-packages/urh/dev/native/lib/usrp.pyx because it changed.
[1/5] Cythonizing /usr/lib64/python3.6/site-packages/urh/dev/native/lib/bladerf.pyx
[2/5] Cythonizing /usr/lib64/python3.6/site-packages/urh/dev/native/lib/hackrf.pyx
[3/5] Cythonizing /usr/lib64/python3.6/site-packages/urh/dev/native/lib/plutosdr.pyx
[4/5] Cythonizing /usr/lib64/python3.6/site-packages/urh/dev/native/lib/rtlsdr.pyx
[5/5] Cythonizing /usr/lib64/python3.6/site-packages/urh/dev/native/lib/usrp.pyx
running clean
removing 'build/lib.linux-x86_64-3.6' (and everything under it)
'build/bdist.linux-x86_64' does not exist -- can't clean it
'build/scripts-3.6' does not exist -- can't clean it
removing 'build'
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Init Process
[INFO::AbstractBaseThread.py::initialize_process] Starting Gnuradio
[DEBUG::AbstractBaseThread.py::initialize_process] /usr/bin/python2 /usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/hackrf_recv.py --samplerate 2000000.0 --freq 314500000.0 --gain 0 --bandwidth 2000000.0 --port 1337 --if-gain 16 --baseband-gain 16
[INFO::AbstractBaseThread.py::initialize_process] Started Gnuradio
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Process Intialized
[INFO::AbstractBaseThread.py::init_recv_socket] Initalizing receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Initalized receive socket
[INFO::AbstractBaseThread.py::init_recv_socket] Trying to get a connection to gnuradio...
[INFO::AbstractBaseThread.py::init_recv_socket] Got connection
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/urh/dev/gr/SpectrumThread.py", line 32, in run
    rcvd += recv(32768)  # Receive Buffer = 32768 Byte
  File "zmq/backend/cython/socket.pyx", line 693, in zmq.backend.cython.socket.Socket.recv (zmq/backend/cython/socket.c:7692)
  File "zmq/backend/cython/socket.pyx", line 727, in zmq.backend.cython.socket.Socket.recv (zmq/backend/cython/socket.c:7469)
  File "zmq/backend/cython/socket.pyx", line 150, in zmq.backend.cython.socket._recv_copy (zmq/backend/cython/socket.c:2446)
  File "zmq/backend/cython/socket.pyx", line 145, in zmq.backend.cython.socket._recv_copy (zmq/backend/cython/socket.c:2353)
  File "zmq/backend/cython/checkrc.pxd", line 19, in zmq.backend.cython.checkrc._check_rc (zmq/backend/cython/socket.c:9908)
zmq.error.Again: Resource temporarily unavailable
Aborted
pentoo-usb ~ #
````

The problematic seems code is where a system-installed urh with python3 by default is trying to run a python2

ZeroChaos-, 2018-12-18:
The specific "silly thing" I was referencing was this:

[DEBUG::AbstractBaseThread.py::initialize_process] /usr/bin/python2 /usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/usrp_recv.py --samplerate 1000000.0 --freq 433920000.0 --gain 50 --bandwidth 1000000.0 --port 1337

calling something installed in the python 3 path but using python2.  Perhaps the issue is on our side and we should be installing urh as python2?

blshkv, 2018-12-18:
No, urh does not support python 2 officially. See setup.py :
print("You need at least Python 3.4

jopohl, 2018-12-18:
Now I see! Here is what is going on:

1. URH requires Python 3.4+ to work, true!
2. GNU Radio, at time of developing the GNU Radio backend, only supported Python 2.7.
3. Some distributions (e.g. Arch Linux) still install GNU Radio as a Python 2.7 lib, thats why the problem _may_ be Gentoo/Pentoo specific (I did not check other distributions, apart from Arch). A solution might be to install GNU Radio as a Python 2 lib on your side? I am, however, not sure if this breaks other applications.

The most easy way to deal with this would be to enter a python 3 interpreter in the options:

[image]

blshkv, 2018-12-18:
@jopohl  we follow each package requirements. In particular, urh was installed as python3 module only and gnuradio as python2 (which seems irrelevant here). 

I think you need to change that menu to:
```
Python3 interpreter = /usr/bin/python3 
```
as a default option

blshkv, 2018-12-18:
```
bash~ $ python2 -c "import gnuradio"
bash ~ $ python3 -c "import gnuradio"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'gnuradio'
```

```
/usr/lib64/python3.6/site-packages/urh
/usr/lib64/python2.7/site-packages/gnuradio
```

jopohl, 2018-12-18:
In that case, you also installed GNU Radio as a python 2 module, so how is it supposed to work overall like requested in this issue? Especially this "silly" line 

```
[DEBUG::AbstractBaseThread.py::initialize_process] /usr/bin/python2 /usr/lib64/python3.6/site-packages/urh/dev/gr/scripts/usrp_recv.py --samplerate 1000000.0 --freq 433920000.0 --gain 50 --bandwidth 1000000.0 --port 1337
```

can't work any other way, because GNU Radio is installed as a Python 2 lib.

blshkv, 2018-12-18:
that exactly the problem. So we either have to wait for GR 3.8 or make URH work with python2.
What should it be?

jopohl, 2018-12-18:
How is this a problem? You call a python 2 script with a python 2 interpreter. The GR backend works on many distributions including Arch and Ubuntu in the way it is now. I don't know what is the exact problem on Gentoo but it has nothing to do with Python 2/3. 

Furthermore, what is the exact reason to use the GR backend in the user's case? Every device (except for FUN Cube dongle) has native support, so there is no need to use the GR backend, because the native one is faster anyway. This is also the reason why the native backend is marked as recommended in the GUI.

ZeroChaos-, 2018-12-18:
it sounds like my assessment of the issue was incorrect, but there is an issue here for sure.

see details in https://github.com/jopohl/urh/issues/587#issuecomment-448038640

jopohl, 2018-12-18:
Oh yeah sure, I was a bit mislead by the python 2/3 topic. I read the comment again, and I missed you are using __spectrum analyzer__, I only tested recording signal. Fix is on the way!

jopohl, 2018-12-18:
Just merged the fix in master. Can you test it on your end? If everything works I will create a new release.

jopohl, 2018-12-19:
New release with included fix is out. I am closing here.

closed 2018-12-19 by jopohl

ZeroChaos-, 2018-12-19:
my user confirmed it's working, thanks!