jopohl/urh #589: USRP: failed to start rx mode

nocomp, 2018-12-21:
##### Expected Behavior

##### Actual Behavior

##### Steps To Reproduce

1. Go to 'FILE'
2. Click on 'Record signal'

4. See error

##### Screenshots
https://imgur.com/a/rHIfwZ6

##### Platform Specifications

 - OS: [e.g. Arch Linux]
 - URH version: [e.g. 2.5.3]
 - Python version: [e.g. 3.6.3]
 - Installed via [msi win 64]
hi
i used to run an old version of URH without any issue.
i ve seen an update, so i ve uninstalled my current version, installed new one, and now , even it manage my usrp b205 as you can see on the screenshot, it never start rx mode.
did i missed something? anything i can do in order to solv it?
thank you for your time
best regards
herve

jopohl, 2018-12-21:
Unfortunately, we do not have any USRP from the B series for testing, so I can only give some abstract support here. The Error comes from the UHD API meaning the USRP could not be opened.

Can you work with your USRP B205 in any other application? If so, can you find out which version of UHD is used by that application?

nocomp, 2018-12-21:
hi,
thx for your time
i had no issues with older urh version, was a 1.5. something and i can use it with gnuradio or wintelive
what could i do in order to help you?

here is the device

-- Loading firmware image: C:\Program Files (x86)\UHD\share\uhd\images\usrp_b200_fw.hex...
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    type: b200
    name: B205i
    serial: 30C7***
    product: B205mini

PS D:\>

nocomp, 2018-12-21:
everything used to work fine with 1.5.* branch, any ideay what changed in URH regarding USRP?

best regards

jopohl, 2018-12-21:
The UHD may be newer (it is bundled with URH), the newer releases come with version 3.12. Can you find out which version is used by GNU Radio (it should print out, when you use the USRP in GNU Radio)

nocomp, 2018-12-22:
good morning @jopohl 
sorry for the delay,
here we go when running grc:

<<< Welcome to GNU Radio Companion 3.7.9.2 >>>

Preferences file: C:\Users\herve/.gnuradio/grc.conf
Block paths:
	C:\Users\herve\.grc_gnuradio (C:\Users\herve/.grc_gnuradio)
	C:\gnuradio\share\gnuradio\grc\blocks (C:\gnuradio\bin\..\share\gnuradio\grc\blocks)

Loading: "D:\sdr\tetra\wintelive\gnuradio_project\wintelive_10channels.grc"
>>> Done

Showing: "D:\sdr\tetra\wintelive\gnuradio_project\wintelive_10channels.grc"

Generating: 'D:\\sdr\\tetra\\wintelive\\gnuradio_project\\top_block.py'

Executing: C:\gnuradio\gr-python27\python.exe -u D:\sdr\tetra\wintelive\gnuradio_project\top_block.py

Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

Using Volk machine: avx2
gr-osmosdr 164a09fc (0.1.5git) gnuradio 3.7.9.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf airspy redpitaya 
-- Loading firmware image: C:\gnuradio\share\uhd\images\usrp_b200_fw.hex...
-- Detected Device: B205mini
-- Loading FPGA image: C:\gnuradio\share\uhd\images\usrp_b205mini_fpga.bin...                                                                                                    1 done
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass

UHD Warning:
    Setting DC offset is not possible on this device.

UHD Warning:
    Setting IQ balance is not possible on this device.
gr::pagesize: no info; setting pagesize = 4096

>>> Done

nocomp, 2018-12-22:
same on 2.4.2

jopohl, 2018-12-22:
GNU Radio seems to use version UHD 3.9.3 in your case. I just had a look at the source code in URH of the usrp backend and found nothing suspicous.

First thing you could try is to just start the recording without pressing the green arrow button, so you perform no device lookup, in this case the lookup is left to the API, maybe this works.

If it does not work, can you install the offical ettus API from [here](http://files.ettus.com/binaries/uhd/latest_release/Windows7/) and see if your USRP gets detected by it? It is the same version URH uses.

nocomp, 2018-12-22:
hello johannes,
thank you for your time
unfortunately, no changes
erro before and after have installed the uhd update.
refreshing or not the device type, no changes, and now even on older version it doesn t work. i ve re installed a r * in another directory, no luck
the update broke something, :'((( dunno what to do. everything works fine with grc or wintelive
and devices, as it was before is recognized

PS C:\Program Files\UHD\bin> .\uhd_find_devices.exe
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_105900; UHD_3.12.0.0-release
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 30C7**
    name: B205i
    product: B205mini
    type: b200

thank you for your helpp

jopohl, 2018-12-22:
Find devices works (this is also used by URH, when you press the green arrow), is there any other program in the driver you could test RX/TX with? Other idea: Can you check the uhd.dll with DependecyWalker and see if you have a missing dependency?

nocomp, 2018-12-22:
Tx rx works fine with grc, i ll check the dll tomorrow morning and let you inow
Thank you very much for your time
Best regards

jopohl, 2018-12-22:
Does GRC also use the newer UHD lib? In the log you posted before it seemed to use and earlier version.

nocomp, 2018-12-23:
hi johannes, here is the log, surely you ll have abetter reading than me

*****************************| System Information |*****************************

Dependency Walker:       2.2.6000 (32-bit)
Operating System:        Microsoft Windows NT/2000/XP/2003/Vista based Home (64-bit)
OS Version:              6.02.9200
Processor:               Intel64 Family 6 Model 94 Stepping 3, GenuineIntel, ~2592MHz
Number of Processors:    8, Mask: 0x000000FF
Computer Name:           LAPTOP-C9KLJGAQ
User Name:               herve
Local Date:              dimanche 23 décembre 2018
Local Time:              10:04:13 Paris, Madrid (GMT+01:00)
OS Language:             0x040C: French (France)
Memory Load:             32%
Physical Memory Total:   2 147 483 647 (2048 MB)
Physical Memory Used:    0
Physical Memory Free:    2 147 483 647
Page File Memory Total:  4 294 967 295
Page File Memory Used:   0
Page File Memory Free:   4 294 967 295
Virtual Memory Total:    2 147 352 576
Virtual Memory Used:     288 690 176
Virtual Memory Free:     1 858 662 400
Page Size:               0x00001000 (4 096)
Allocation Granularity:  0x00010000 (65 536)
Min. App. Address:       0x00010000 (65 536)
Max. App. Address:       0x7FFEFFFF (2 147 418 111)

********************************| Search Order |********************************
*                                                                              *
* Legend: F  File                     E  Error (path not valid)                *
*                                                                              *
********************************************************************************

Side-by-Side components (Windows 2000/XP/2003/Vista/+)
The system's "KnownDLLs" list
   [F ] c:\windows\system32\ADVAPI32.DLL
   [F ] c:\windows\system32\BCRYPTPRIMITIVES.DLL
   [F ] c:\windows\system32\CFGMGR32.DLL
   [F ] c:\windows\system32\CLBCATQ.DLL
   [F ] c:\windows\system32\COMBASE.DLL
   [F ] c:\windows\system32\COMCTL32.DLL
   [F ] c:\windows\system32\COMDLG32.DLL
   [F ] c:\windows\system32\COML2.DLL
   [F ] c:\windows\system32\CRYPT32.DLL
   [F ] c:\windows\system32\DIFXAPI.DLL
   [F ] c:\windows\system32\FLTLIB.DLL
   [F ] c:\windows\system32\GDI32.DLL
   [F ] c:\windows\system32\GDI32FULL.DLL
   [F ] c:\windows\system32\GDIPLUS.DLL
   [F ] c:\windows\system32\IMAGEHLP.DLL
   [F ] c:\windows\system32\IMM32.DLL
   [F ] c:\windows\system32\KERNEL.APPCORE.DLL
   [F ] c:\windows\system32\KERNEL32.DLL
   [F ] c:\windows\system32\KERNELBASE.DLL
   [F ] c:\windows\system32\MSASN1.DLL
   [F ] c:\windows\system32\MSCTF.DLL
   [F ] c:\windows\system32\MSVCP_WIN.DLL
   [F ] c:\windows\system32\MSVCRT.DLL
   [F ] c:\windows\system32\NORMALIZ.DLL
   [F ] c:\windows\system32\NSI.DLL
   [F ] c:\windows\system32\NTDLL.DLL
   [F ] c:\windows\system32\NTDLL.DLL
   [F ] c:\windows\system32\OLE32.DLL
   [F ] c:\windows\system32\OLEAUT32.DLL
   [F ] c:\windows\system32\POWRPROF.DLL
   [F ] c:\windows\system32\PROFAPI.DLL
   [F ] c:\windows\system32\PSAPI.DLL
   [F ] c:\windows\system32\RPCRT4.DLL
[... 1370 lines cut ...]
[D  ]  DSPARSE.DLL                                                   12/04/2018 00:34  03/04/2084 22:37     24 064  A      0x00010E52     0x00010E52     x86  Console    CV,Unknown  0x--------10000000  Unknown      0x0000B000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    10.0
[D  ]  IMAGEHLP.DLL                                                  12/04/2018 00:34  13/05/2054 14:36     95 584  A      0x0001F597     0x0001F597     x86  Console    CV,Unknown  0x--------10000000  Unknown      0x00019000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    6.1
[D  ]  IPHLPAPI.DLL                                                  12/04/2018 00:34  10/10/2090 13:42    190 136  A      0x0003C2A8     0x0003C2A8     x86  Console    CV,Unknown  0x--------10000000  Unknown      0x00030000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    10.0
[D  ]  MSASN1.DLL                                                    12/04/2018 00:34  09/06/2055 10:32     49 656  A      0x0000D5A1     0x0000D5A1     x86  GUI        CV,Unknown  0x--------10000000  Unknown      0x0000E000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    10.0
[D  ]  MSIMG32.DLL                                                   12/04/2018 00:34  16/08/2043 22:36      7 168  A      0x00001C57     0x00001C57     x86  Console    CV,Unknown  0x--------5B800000  Unknown      0x00006000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    6.0
[D  ]  NCRYPT.DLL                                                    12/04/2018 00:34  29/10/2037 02:49    118 872  A      0x0002118D     0x0002118D     x86  Console    CV,Unknown  0x--------10000000  Unknown      0x0001F000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    10.0
[D  ]  NSI.DLL                                                       12/04/2018 00:34  22/04/1998 22:37     20 216  A      0x0000DC07     0x0000DC07     x86  Console    CV,Unknown  0x--------10000000  Unknown      0x00007000    Not Loaded  10.0.17134.1    10.0.17134.1    10.0       14.12       10.0    10.0
[D  ]  NTASN1.DLL                                                    12/04/2018 00:34  21/04/1979 08:45    178 256  A      0x00038950     0x00038950     x86  GUI        CV,Unknown  0x--------10000000  Unknown      0x0002D000    Not Loaded  10.0.17134.1    10.0.17134.1  [... cut ...]
************************************| Log |*************************************

Error: At least one required implicit or forwarded dependency was not found.
Error: At least one module has an unresolved import due to a missing export function in an implicitly dependent module.
Error: Modules with different CPU types were found.
Warning: At least one delay-load dependency module was not found.
Warning: At least one module has an unresolved import due to a missing export function in a delay-load dependent module.

nocomp, 2018-12-23:
here si the grc launchin log

<<< Welcome to GNU Radio Companion 3.7.9.2 >>>

Preferences file: C:\Users\herve/.gnuradio/grc.conf
Block paths:
	C:\Users\herve\.grc_gnuradio (C:\Users\herve/.grc_gnuradio)
	C:\gnuradio\share\gnuradio\grc\blocks (C:\gnuradio\bin\..\share\gnuradio\grc\blocks)

Loading: "D:\sdr\tetra\wintelive\gnuradio_project\wintelive_10channels.grc"
>>> Done

Showing: "D:\sdr\tetra\wintelive\gnuradio_project\wintelive_10channels.grc"

Generating: 'D:\\sdr\\tetra\\wintelive\\gnuradio_project\\top_block.py'

Executing: C:\gnuradio\gr-python27\python.exe -u D:\sdr\tetra\wintelive\gnuradio_project\top_block.py

Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

Using Volk machine: avx2
gr-osmosdr 164a09fc (0.1.5git) gnuradio 3.7.9.2
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf airspy redpitaya 
-- Loading firmware image: C:\gnuradio\share\uhd\images\usrp_b200_fw.hex...
-- Detected Device: B205mini
-- Loading FPGA image: C:\gnuradio\share\uhd\images\usrp_b205mini_fpga.bin...                                                                                                    1 done
-- Operating over USB 2.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Using subdev spec 'A:A'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass

UHD Warning:
    Setting DC offset is not possible on this device.

UHD Warning:
    Setting IQ balance is not possible on this device.
gr::pagesize: no info; setting pagesize = 4096

jopohl, 2018-12-23:
You see GNU Radio uses an earlier version of UHD:
> Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

Can you post a screenshot of Dependency Walker anaylzing the DLL? That it is easier to see, however, it says you miss some DLLs so that might be the problem.

nocomp, 2018-12-23:
hello
i ve did a sc, but i doubt you can see it all, so i ve uploaded the walker file too, might be more usefull
https://drive.google.com/file/d/1uXfj7uWXt1-Cru62hxVbM8zmCg4faZ_o/view?usp=sharing

thxx for your time and help, truelly appreciate

nocomp, 2018-12-23:
just tried the uhd.dll from grc and i have the same result
the ntdll.dll exist in system32/ where else should i copy it?

nocomp, 2018-12-23:
here is a csv file output in case it helps
https://drive.google.com/file/d/1EiMEEh-VHAUPYXK5AcNpWpg3P2svNP8f/view?usp=sharing

nocomp, 2018-12-23:
i ve just uninstalled all urh versions, runned ccleaner, reinstall last version, same error 44 :/

nocomp, 2018-12-23:
any idea how i can fix the missing dll issue even they are not missing?

nocomp, 2018-12-23:
it seems that dependancy walker gives lots of false positive https://ofekshilon.com/2016/03/27/on-api-ms-win-xxxxx-dll-and-other-dependency-walker-glitches/

nocomp, 2018-12-24:
hello
tryin to install on linux, but i guess i am coarsed, even all deps were installed, followed your tut.
here is what i come with:
(kali-rolling)nocomp@localhost:~$ sudo urh
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 7, in <module>
    from urh.main import main
  File "/usr/local/lib/python3.6/dist-packages/urh/main.py", line 9, in <module>
    from PyQt5.QtGui import QPalette, QIcon, QColor
ImportError: libGLESv2.so.2: cannot open shared object file: No such file or directory
(kali-rolling)nocomp@localhost:~$ 

am i close?

thxx for helping

nocomp, 2018-12-24:
(kali-rolling)nocomp@localhost:~$ ldconfig -p | grep libGLESv2
	libGLESv2.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libGLESv2.so
(kali-rolling)nocomp@localhost:~$

nocomp, 2018-12-24:
i guess a sim link is issing, get same error when launching gqrx

nocomp, 2018-12-24:
finally got it almost working!!!!!
i ve uninstalled and clean gnuradio and all the sdr app and pip and urh, and reinstalled from scratch
usrp is enable, but this is the log when i do a spectrum analyzer, rx still not working, what is missing?

[32;1m[INFO] [UHD] [39;0mlinux; GNU C++ version 8.2.0; Boost_106200; UHD_3.13.0.2-3
[32;1m[INFO] [B200] [39;0mDetected Device: B205mini
[32;1m[INFO] [B200] [39;0mOperating over USB 2.
[32;1m[INFO] [B200] [39;0mInitialize CODEC control...
[32;1m[INFO] [B200] [39;0mInitialize Radio control...
[32;1m[INFO] [B200] [39;0mPerforming register loopback test... 
[32;1m[INFO] [B200] [39;0mRegister loopback test passed
[32;1m[INFO] [B200] [39;0mSetting master clock rate selection to 'automatic'.
[32;1m[INFO] [B200] [39;0mAsking for clock rate 16.000000 MHz... 
[32;1m[INFO] [B200] [39;0mActually got clock rate 16.000000 MHz.
[32;1m[INFO] [B200] [39;0mAsking for clock rate 32.000000 MHz... 
[32;1m[INFO] [B200] [39;0mActually got clock rate 32.000000 MHz.

nocomp, 2018-12-25:
hello
no error at all
:'(

lesson #1: if it works, don t upgrade...

back to windows, need to figure out why nothing is working anymore since upgrade

nocomp, 2018-12-25:
i ve uninstall all win versions and reinstalled 1.8.15 wich seems the one i was running
now i have this, might explain it all
https://imgur.com/w4fekfr

nocomp, 2018-12-25:
GOT IT WORKING !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

changed backend to gnuradio and no more error.

i guess i should stay to 1.8.15 or is it worth to try to update?

jopohl, 2018-12-25:
GNU Radio backend should keep working in recent version, as long as you don't change your GNU Radio installation.

nocomp, 2018-12-25:
damm what if i tell you i am scare to upgrade and break it all ... again ...

andynoack, 2018-12-25:
Make a snapshot and give it a try :) Because a lot of things have improved since 1.8.15! Anyone has contact to Ettus and can arrange that we get our hands on B20x's?

nocomp, 2019-07-14:
hi,
sorry again to re open this topic, can t find anywhere the device argument to add in urh when using b205 mini in usb.
thxx for your help

jopohl, 2019-07-14:
What you mean? There is a auto detect button for this in recent versions.

nocomp, 2019-07-14:
I run an older version with no auto detect

jopohl, 2019-07-14:
Are you searching for this? https://files.ettus.com/manual/page_identification.html

Nevertheless, updating URH would be the preferred option.

nocomp, 2019-07-14:
Oki, i ll follow your recomandation.
Thxx for your help

nocomp, 2019-07-15:
hi ,
just downloaded the last version and installed it after have uninstall the older one.

it see the b205 type=b200,name=B205i,serial=30C7E1D,product=B205mini but either i run a sniff or a record, it doesn t show up anything, doesn t record a thing
samples captured remains at 0 as for other like buffer received, size and time.
running it as admin doesn t do a change, am i missing something?
thxx for your help

best regards

nocomp, 2019-07-15:
here are some extra logs
USRP-OPEN (type=b200,name=B205i,serial=30C7E1D,product=B205mini): Error Code: 11 (11)
USRP: failed to start rx mode

hope it can help

nocomp, 2019-07-16:
hi
so this is the error as usual with last version
USRP-OPEN (type=b200,name=B205i,serial=30C7E1D,product=B205mini): Error Code: 44 (44)
USRP: failed to start rx mode

usrp works fine with gnuradio , and as you can see device is detected by auto detect feature.
what am i missing then?
thx for your time

jopohl, 2019-07-16:
Maybe you need to update the USB driver for the B205?

nocomp, 2019-07-16:
Will try, but it was xorking with older version of urh

nocomp, 2019-08-13:
hi,
sorry for the late reply i ve been away.
i ve removed everything as follow:

removed uhd
removed urh
removed gnuradio

then i ve download all latest versions and then 
reinstalled urh
reinstalled gnuradio
reinstalled urh

and still the same error:
USRP-OPEN (type=b200,name=B205i,serial=30C7E1D,product=B205mini): Error Code: 44 (44)
USRP: failed to start rx mode

regarding usrp install:
PS C:\Users\herve> & 'C:\Program Files\PothosSDR\bin\uhd_usrp_probe.exe'
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.004-0-g2b5a88bb

-- Detected Device: B205mini
-- Loading FPGA image: C:\Program Files\UHD\share\uhd\images\usrp_b205mini_fpga.bin... done
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz...
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B205mini
|   |   revision: 1
|   |   product: 30522
|   |   serial: 30C7E1D
|   |   name: B205i
|   |   FW Version: 8.0
|   |   FPGA Version: 4.0
|   |
|   |   Time sources: none, internal, external
|   |   Clock sources: internal, external
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B205mini RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.3 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B205mini TX dual DAC
|   |   |   |   Gain Elements: None

PS C:\Users\herve>

i am despaired to get this working, any hints?
thxx for you time

jopohl, 2019-08-13:
I did some fixes regarding the B series but did not make a new release yet. You can either run from source or download a fresh release [here](https://dev.azure.com/jopohl/7123cbff-1b7f-4129-8d40-44a879486bca/_apis/build/builds/802/artifacts?artifactName=dist&api-version=5.2-preview.5&%24format=zip).

I gonna make a new release on monday.

nocomp, 2019-08-13:
damm, i was so happy to see a drop menu for antenna choice and everything, but same error
USRP-OPEN (type=b200,name=B205i,serial=30C7E1D,product=B205mini): Error Code: 44 (44)
USRP: Could not retrieve more detailed error message from device.
USRP: failed to start rx mode

subdevice category is empty
well if i can help you out debugging, here to help!

thxx for your time

jopohl, 2019-08-13:
Is your firmware up to date?

nocomp, 2019-08-13:
isn t downloading the latest image everytime?

jopohl, 2019-08-13:
Is it? I do not have a B device so I do not know. I had a user with a similar issue on a X series device and updating the firmware did the trick.

nocomp, 2019-08-14:
doesn t seems we can update the firmware.
just tried on kali works like a charm, need sudo, maybe it s a permission issue too with windows for have access to usb, but even run as admin, same error.
thxx for your time, truelly appreciate

herve

jopohl, 2019-08-14:
Ohh Windows and USB! This is a story for itself. You may be lucky when you try a different USB port but since it works on Linux I agree there is some Windows USB voodoo going on.