jopohl/urh #590: How to run?

OZ1SEJ, 2018-12-25:
##### Expected Behavior
I expected some kind of indication as to how to actually run the program.

##### Actual Behavior
In Ubuntu Apps, typing "urh" returns "Nothing found." Typing "urh &" in a terminal returns "not found". Typing "python3 urh" returns "No such file or directory."

##### Steps To Reproduce
I installed URH with pip3 install urh.

##### Screenshots
$ pip3 install urh
Collecting urh
  Downloading https://files.pythonhosted.org/packages/a0/5b/5e5d24904d4141af8e8733cc56a0e26ab9ee420fcde741432e2dcb492c59/urh-2.5.3.tar.gz (3.2MB)
    100% |████████████████████████████████| 3.2MB 536kB/s 
Collecting cython (from urh)
  Using cached https://files.pythonhosted.org/packages/39/b1/2acbf92bb3c817dc99a7588a6196629a0490b3f940b672136aa4d09f91ea/Cython-0.29.2-cp36-cp36m-manylinux1_x86_64.whl
Collecting numpy (from urh)
  Downloading https://files.pythonhosted.org/packages/ff/7f/9d804d2348471c67a7d8b5f84f9bc59fd1cefa148986f2b74552f8573555/numpy-1.15.4-cp36-cp36m-manylinux1_x86_64.whl (13.9MB)
    100% |████████████████████████████████| 13.9MB 136kB/s 
Collecting psutil (from urh)
  Downloading https://files.pythonhosted.org/packages/e3/58/0eae6e4466e5abf779d7e2b71fac7fba5f59e00ea36ddb3ed690419ccb0f/psutil-5.4.8.tar.gz (422kB)
    100% |████████████████████████████████| 430kB 1.8MB/s 
Collecting pyqt5 (from urh)
  Downloading https://files.pythonhosted.org/packages/d4/bf/d884da8e2f7096d201c891d515eb6813a8e85df5eb6f5e12e867bf1d831c/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (117.8MB)
    100% |████████████████████████████████| 117.8MB 15kB/s 
Collecting pyzmq (from urh)
  Downloading https://files.pythonhosted.org/packages/48/93/59592cb294761aaa40589b544eaa5175446d687ff95beeeb666de60f3274/pyzmq-17.1.2-cp36-cp36m-manylinux1_x86_64.whl (998kB)
    100% |████████████████████████████████| 1.0MB 673kB/s 
Collecting PyQt5_sip<4.20,>=4.19.11 (from pyqt5->urh)
  Downloading https://files.pythonhosted.org/packages/46/ba/30995383332fd46e4ee59a405bd75ae8ee099001360e7a680f1685985a49/PyQt5_sip-4.19.13-cp36-cp36m-manylinux1_x86_64.whl (66kB)
    100% |████████████████████████████████| 71kB 4.4MB/s 
Building wheels for collected packages: urh, psutil
  Running setup.py bdist_wheel for urh ... done
  Stored in directory: /home/steen/.cache/pip/wheels/69/41/e6/a7f00c82eeda7ff0b1e8b981e475c4b140f5d7ee0d560c18d7
  Running setup.py bdist_wheel for psutil ... done
  Stored in directory: /home/steen/.cache/pip/wheels/d2/71/40/9c6993129f8cda369d0f21c46a13a6adab7fb1664fe6512551
Successfully built urh psutil
Installing collected packages: cython, numpy, psutil, PyQt5-sip, pyqt5, pyzmq, urh
Successfully installed PyQt5-sip-4.19.13 cython-0.29.2 numpy-1.15.4 psutil-5.4.8 pyqt5-5.11.3 pyzmq-17.1.2 urh-2.5.3

##### Platform Specifications
 - OS: Ubuntu 18.04
 - URH version: 2.5.3
 - Python version: 3.6.7
 - Installed via pip3

OZ1SEJ, 2018-12-25:
Also, it doesn't say in the manual how to actually run URH.

jopohl, 2018-12-25:
Since you used ``` pip3 install urh ``` to install it (not sudo) it got installed to a local bin directory which seems not to be in your path. You could install it system wide with ``` sudo pip3 install urh ``` or just use e.g. ``` locate ``` to find out where it got installed to and add this directory to your PATH.

OZ1SEJ, 2018-12-25:
Thanks! I installed it with sudo, but now when I run it, I get an error:

```
$ urh &
[2] 21581
$ Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/home/steen/.local/lib/python3.6/site-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "__init__.pxd", line 918, in init urh.cythonext.signal_functions
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject
```

jopohl, 2018-12-25:
Seems like the numpy from PyPI is incompatible with the rest of Ubuntu 18.04. You should uninstall that (pip3 uinstall numpy) and install the one from the repos, just as given in the [install instructions](https://github.com/jopohl/urh#generic-way-with-pip-ubuntudebian).