jopohl/urh #592: Python ValueError when launching URH

Christopher-A, 2019-01-02:
When I launch URH I receive the following error:
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "__init__.pxd", line 918, in init urh.cythonext.signal_functions
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject

I've tried different versions of numpy with no success.

##### Platform Specifications
 - OS: MacOS 10.14.1
 - URH version: 2.5.3
 - Python version: Python 3.7.2
 - Installed via pip3

jopohl, 2019-01-02:
Quoting myself from [this thread](https://github.com/jopohl/urh/issues/273):

Can't reproduce. Here is what I did:
1. Create a fresh OSX 10.14 VM.
2. Download Python from official website.
3. ``` sudo python -m pip3 install urh ```
 a. Click through all these flashing up fancy OSX dialogs
 b. run ``` sudo python3 -m pip install urh ``` again
4. Start URH -> works

I suppose you have some varying python libraries on your system. OSX is very good at messing this up. Is it possible you installed some python libraries with brew and others with pip? This may explain the issue.

jopohl, 2019-01-02:
Just a shot in the dark: Is your cython up-to-date? 
```
python3 -m pip install --upgrade cython
```

Christopher-A, 2019-01-04:
Per that thread, downgrading to 2.5.1 has worked for me.

closed 2019-01-04 by Christopher-A