jopohl/urh #601: Frequent crash outs in Windows 32bit

mr-sneezy, 2019-01-14:
##### Expected Behavior
 Expected less frequent program crashes. 

##### Actual Behavior
Crashes in Windows version quite regular, usually just after a file save, sometimes after closing a window.
##### Steps To Reproduce
Saving files, closing a window.

##### Platform Specifications

 - OS: Win10
 - URH version: [e.g. 2.5.4]
 - Installed via MSI

I'd like to help resolve the crashes. What can I do to capture crash reporting ?
URH is working well enough for me to make use of it though between crashes...

jopohl, 2019-01-14:
Please post the output of ``` %TEMP%\urh.log ``` after a crash.

jopohl, 2019-01-14:
I just created added a debug exe to the release.
1. Please download the release from [here](https://www.dropbox.com/s/t76m9l5k2eql59e/Universal%20Radio%20Hacker-2.5.4-win32.msi?dl=0).
2. Install it.
3. Open your installation path (default ```C:\Program Files\Universal Radio Hacker```)
4. Run ``` urh_debug.exe ``` from a terminal. Ensure, you are in the directory from step 3.
5. Now use URH as usual until you can reproduce a crash.
6. Post the output from the terminal here.

mr-sneezy, 2019-01-14:
Done. I recorded two chirps from my new weather station via an RTL-SDR dongle, then saved the file, then tried to close the Recorder window, initially it seems that Interpret loads something behind the Recorder window, then both windows disappear (crash out). File is saved, I can find it next time I open URH.

C:\Program Files (x86)\Universal Radio Hacker>urh_debug.exe

[INFO::SettingsProxy.py::get_receive_buffer_size] Correcting buffer size to 134217727
[INFO::SettingsProxy.py::get_receive_buffer_size] Try to allocate receive buffer with size 1.074GB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode

Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success
E[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433890000.0: Success
xact sample rate is: 1000000.026491 Hz
[INFO::Device.py::log_retcode] RTLSDR-SET_SAMPLE_RATE to 1000000.0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_BANDWIDTH to 1000000.0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY_CORRECTION to 1: Success
Disabled direct sampling mode
[INFO::Device.py::log_retcode] RTLSDR-SET_DIRECT_SAMPLING_MODE to 0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_RF_GAIN to 37: Success
[INFO::Device.py::log_retcode] RTLSDR-RESET_BUFFER: Success
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[DEBUG::RTLSDR.py::shutdown_device] RTLSDR: closing device
[INFO::Device.py::log_retcode] RTLSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::read_device_messages] Exiting read device message thread due to
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode

Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success
E[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433890000.0: Success
xact sample rate is: 1000000.026491 Hz
[INFO::Device.py::log_retcode] RTLSDR-SET_SAMPLE_RATE to 1000000.0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_BANDWIDTH to 1000000.0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY_CORRECTION to 1: Success
Disabled direct sampling mode
[INFO::Device.py::log_retcode] RTLSDR-SET_DIRECT_SAMPLING_MODE to 0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_RF_GAIN to 37: Success
[INFO::Device.py::log_retcode] RTLSDR-RESET_BUFFER: Success
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[DEBUG::RTLSDR.py::shutdown_device] RTLSDR: closing device
[INFO::Device.py::log_retcode] RTLSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::read_device_messages] Exiting read device message thread due to
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialog.py::closeEvent] Device stopped successfully.
Traceback (most recent call last):
  File ".\src\urh\controller\MainController.py", line 720, in on_signals_recorded
  File ".\src\urh\controller\MainController.py", line 288, in add_signalfile
  File ".\src\urh\controller\MainController.py", line 309, in add_signal
  File ".\src\urh\controller\widgets\SignalFrame.py", line 1158, in refresh
  File ".\src\urh\controller\widgets\SignalFrame.py", line 519, in show_protocol
  File ".\src\urh\controller\widgets\SignalFrame.py", line 506, in update_protocol
  File ".\src\urh\signalprocessing\ProtocolAnalyzer.py", line 226, in get_protocol_from_signal
  File "src\urh\cythonext\signal_functions.pyx", line 343, in urh.cythonext.signal_functions.grab_pulse_lens
  File "src\urh\cythonext\signal_functions.pyx", line 363, in urh.cythonext.signal_functions.grab_pulse_lens
MemoryError

C:\Program Files (x86)\Universal Radio Hacker>

jopohl, 2019-01-14:
How much RAM do you have in your system? You are running out of memory.

jopohl, 2019-01-14:
You could try lowering this setting here and set it to 20% or so:

[image]

mr-sneezy, 2019-01-15:
OK will try that.
I have 16Gb RAM in the system (15Gb usable). Running Win10 64bit, Intel HD onboard graphics.

mr-sneezy, 2019-01-15:
No, changing RAM use to 20% made no difference. First test failed when closing Recorder after file save.
What else can I log ?

jopohl, 2019-01-15:
>OK will try that.
I have 16Gb RAM in the system (15Gb usable). Running Win10 64bit, Intel HD onboard graphics.

Why are you using the 32 bit version of URH then?

jopohl, 2019-01-15:
> No, changing RAM use to 20% made no difference. First test failed when closing Recorder after file save.
What else can I log ?

Please set it to 5% or so. Since you have 16 GB of RAM, but run a 32 Bit Python such things can be expected. Have you tried the 64 bit version?

jopohl, 2019-01-15:
I think I found the cause of everything. Should be fixed in next release. For now, you can simply uncheck ``` Edit ``` -> ``` Auto detect signals on loading ``` to fix the crashes.

mr-sneezy, 2019-01-15:
OK, will try that in the morning.
Re the 64bit version and using the 32bit. I am guessing now, that the version ending in 'amd64.msi' is not a special AMD only version then like I thought (I have an Intel i3 CPU) ?

jopohl, 2019-01-15:
> Re the 64bit version and using the 32bit. I am guessing now, that the version ending in 'amd64.msi' is not a special AMD only version then like I thought (I have an Intel i3 CPU) ?

Correct.

mr-sneezy, 2019-01-16:
OK that solution works. 
"Edit -> Auto detect signals on loading to fix the crashes."
Have not crashed after a few recordings and saves now.
Also now using 64bit version.

jopohl, 2019-01-16:
Cool! In 64 bit version you can also leave " Auto detect signals on loading to fix the crashes" checked, because the crashes only affected 32 bit version.

Furthermore, I fixed the underlying reason for 32 bit crashes, so I will close here.

closed 2019-01-16 by jopohl